borglab/gtsam #1674: cmake fails with -DGTSAM_UNSTABLE_INSTALL_MATLAB_TOOLBOX

etamarlu, 2023-11-21:
## Description

## Steps to reproduce
The following works as excepted:
```
cmake -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=/home/etamar/Desktop/toolbox  -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON  -
```
when I add DGTSAM_UNSTABLE_INSTALL_MATLAB_TOOLBOX =ON in the end it returns an error

```
cmake -DGTSAM_TOOLBOX_INSTALL_PATH:PATH=/home/etamar/Desktop/toolbox  -DGTSAM_BUILD_UNSTABLE=ON -DGTSAM_INSTALL_MATLAB_TOOLBOX=ON  -DGTSAM_UNSTABLE_INSTALL_MATLAB_TOOLBOX=ON  ..
```
1. 
2.

## Expected behavior

## Environment

## Additional information
The log/error:
```
-- GTSAM is a shared library due to GTSAM_FORCE_SHARED_LIB
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Eigen version: 3.4.0
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
CMake Warning at cmake/HandleTBB.cmake:4 (find_package):
  By not providing "FindTBB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TBB", but
  CMake did not find one.

  Could not find a package configuration file provided by "TBB" (requested
  version 4.4) with any of the following names:

    TBBConfig.cmake
    tbb-config.cmake

  Add the installation prefix of "TBB" to CMAKE_PREFIX_PATH or set "TBB_DIR"
  to a directory containing one of the above files.  If "TBB" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:79 (include)

-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3a0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3a0
-- Install prefix: /usr/local
-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
-- Building wrap module gtsam
-- Installing Matlab Toolbox to /home/etamar/Desktop/toolbox
CMake Error at matlab/CMakeLists.txt:101 (matlab_wrap):
  matlab_wrap Function invoked with incorrect arguments for function named:
  matlab_wrap

-- Wrote /home/etamar/Desktop/gtsam-develop/build/GTSAMConfig.cmake
-- Wrote /home/etamar/Desktop/gtsam-develop/build/GTSAM_UNSTABLEConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 11.4.0
--  CMake version                                    : 3.22.1
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/gmake
-- Build flags                                               
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Enabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Enabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            :  -O3 -DNDEBUG
--  Enable Boost serialization                       : ON
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_17
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  Use System Eigen                                 : OFF (Using version: 3.4.0)
--  Use System Metis                                 : OFF
--  Using Boost version                              : 1.74.0
--  Use Intel TBB                                    : TBB not found
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : STL
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Build with Memory Sanitizer                      : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Enable branch merging in DecisionTree            : Enabled
--  Allow features deprecated in GTSAM 4.3           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Enabled
--  MATLAB root                                      : /usr/local/MATLAB/R2023b
--  MEX binary                                       : /usr/local/MATLAB/R2023b/bin/glnxa64/mex
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
CMake Warning at cmake/HandleFinalChecks.cmake:3 (message):
  TBB 4.4 or newer was not found - this is ok, but note that GTSAM
  parallelization will be disabled.  Set GTSAM_WITH_TBB to 'Off' to avoid
  this warning.
Call Stack (most recent call first):
  CMakeLists.txt:159 (include)

-- Configuring incomplete, errors occurred!
See also "/home/etamar/Desktop/gtsam-develop/build/CMakeFiles/CMakeOutput.log".
See also "/home/etamar/Desktop/gtsam-develop/build/CMakeFiles/CMakeError.log".
```

etamarlu, 2023-11-27:
There was a missing field in the "matlab_wrap" call in 'gtsam/matlab/CMakeLists.txt', line 101.

The following change fix it:
```
 matlab_wrap(${GTSAM_SOURCE_DIR}/gtsam_unstable/gtsam_unstable.i "gtsam_unstable"
              "${GTSAM_ADDITIONAL_LIBRARIES}" "" "${mexFlags}" "${ignore}")
```
to 
```
 matlab_wrap(${GTSAM_SOURCE_DIR}/gtsam_unstable/gtsam_unstable.i "gtsam_unstable"
              "${GTSAM_ADDITIONAL_LIBRARIES}" "" "${mexFlags}" "${ignore}" "${GTSAM_ENABLE_BOOST_SERIALIZATION}")
```

ProfFan, 2023-11-27:
@etamarlu Could you submit a PR? Thank you!

closed 2023-11-27 by etamarlu