borglab/gtsam #1682: python error

LiXinghui-666, 2023-12-05:
When I finished compiling, I ran pip install ./python in the build directory and it showed that gtsam was installed normally. But when I tried it, the following error occurred: 
```python
Python 3.8.18 (default, Sep 11 2023, 13:40:15) 
[GCC 11.2.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gtsam
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/data2/conda/miniconda3/envs/nerf-slam/lib/python3.8/site-packages/gtsam/__init__.py", line 7, in <module>
    from gtsam import gtsam, utils
ImportError: cannot import name 'gtsam' from partially initialized module 'gtsam' (most likely due to a circular import) (/data2/conda/miniconda3/envs/nerf-slam/lib/python3.8/site-packages/gtsam/__init__.py)
```
How can I solve it?

pie-leroy, 2023-12-19:
Hello, same error for me

varunagrawal, 2023-12-29:
If you're running `python` inside the build directory, it will pick up the local directory `gtsam` instead of the gtsam python module. I recommend running `python` from another directory (preferably the home directory).

I highly recommend understanding the import process in Python to avoid issues like this in the future.

varunagrawal, 2024-01-05:
Closing due to lack of a response. Please reopen if this is still an issue.

closed 2024-01-05 by varunagrawal

liuhj86, 2024-03-20:
same issue.

varunagrawal, 2024-03-20:
Please run `make python-install` from the build directory and try again. This seems to be an issue with the way installation is being performed on a user-basis.