borglab/gtsam #1683: Is the NonlinearFactorGraph addPrior function deprecated?

zhao-zhibo, 2023-12-06:
I saw that the addPrior function in the header file of NonlinearFactorGraph has been deprecated, but the cpp in gtsam.pdf and other examples given still use the addPrior function, which causes the example to fail to compile. Please correct it.

varunagrawal, 2024-01-02:
Where do you see addPrior being deprecated? I just looked at NonlinearFactorGraph.h and don't see any deprecation markings.

varunagrawal, 2024-01-03:
Closing this issue since it is invalid.

closed 2024-01-03 by varunagrawal

zhao-zhibo, 2024-01-04:
@varunagrawal Because when I use it, the addprior function is not available, and one yuan factor cannot be added normally, but I already contain nonlinearfactorgraph.h head files. It will not report an error if you replace it with the ADD function. The code is shown below.The code comes from this CPP  [testGtsam](https://github.com/zhao-zhibo/test_gtsam/blob/main/src/testGtsam.cpp).
```
gtsam::NonlinearFactorGraph graph;
auto priorNoise = gtsam::noiseModel::Diagonal::Sigmas(gtsam::Vector3(0.3, 0.3, 0.1));
// graph.addPrior(1, gtsam::Pose2(0,0,0), priorNoise); addPrior函数已经被删除了，用add函数即可
 graph.add(gtsam::PriorFactor<gtsam::Pose2>(1, gtsam::Pose2(0,0,0), priorNoise));
```