borglab/gtsam #1689: compile gtsam on the ios using conan failing with a bug

11janson, 2023-12-12:
## Description

[https://github.com/conan-io/conan-center-index/issues/21725](url)
https://github.com/conan-io/conan-center-index/issues/21725
## Steps to reproduce

1. 
2.

## Expected behavior

## Environment

## Additional information

ProfFan, 2023-12-14:
Please post your complete issue here, thanks.

11janson, 2023-12-19:
> Please post your complete issue here, thanks.

hello，I have complete it.

varunagrawal, 2023-12-29:
I believe the issue here is that the `system` function is not available on iOS. From the MacOSX.sdk `stdlib.h`

```c
__swift_unavailable("Use posix_spawn APIs or NSTask instead. (On iOS, process spawning is unavailable.)")
__API_AVAILABLE(macos(10.0)) __IOS_PROHIBITED
__WATCHOS_PROHIBITED __TVOS_PROHIBITED
int	 system(const char *) __DARWIN_ALIAS_C(system);
```

A fix for you could be to conditionally compile the `system` call in `DecisionTree-inl.h`. That is the only place it is used. We would welcome a PR and I can help land it. It should be really simple to fix.

11janson, 2024-01-02:
> I believe the issue here is that the `system` function is not available on iOS. From the MacOSX.sdk `stdlib.h`
>
> ```c
> __swift_unavailable("Use posix_spawn APIs or NSTask instead. (On iOS, process [...]

Thanks. I will push a PR a few days later.

varunagrawal, 2024-02-09:
@11janson since we don't officially support iOS, your best bet moving forward is to fork GTSAM and either remove that function or conditionally compile it for your purposes. Closing this issue.

closed 2024-02-09 by varunagrawal

LockedThread, 2024-03-09:
> Thanks. I will push a PR a few days later.

Did you push a PR for this?