borglab/gtsam #1693: GTSAM detect wrong data type with Point3 in *windows*

etamarlu, 2023-12-17:
## Description
I built GTSAM-develop with Matlab toolbox in windows (it required applying some patches from [feedstock](https://github.com/conda-forge/gtsam-feedstock/tree/main/recipe), but the same errors appears when I use their GTSAM 4.2 which they officially support)

when running any trying to insert a Vector (NX1) to Values an error will appear when calling the solver (I tried ISAM2).
I tracked it down to [NonlinearFactor.h.h](https://github.com/borglab/gtsam/blob/9d70605d488968eb766874d210b3f734149f3b66/gtsam/nonlinear/NonlinearFactor.h#L700) at lines 707 or 710, when calling 
```
x.at<ValueTypes>(keys_[Indices])
```

For some reason there is a discrepancy between ValueType and the value type saved.

I found similar problems in the past, ie, [1](https://github.com/borglab/gtsam/issues/1519), [2](https://github.com/borglab/gtsam/issues/1266)

I saw that a possible reason was the order of overloaded functions, but I couldn't find why it happens here.

## Steps to reproduce

1. 
2.

## Expected behavior

## Environment

## Additional information

ProfFan, 2024-01-02:
This could be related to #1685, could you take a look? Thank you very much!

talregev, 2024-01-15:
> ```
> Exception from gtsam:
> Attempting to retrieve value with key "v3", type stored in Values is class gtsam::GenericValue<class
> Eigen::Matrix<double,-1,1,0,-1,1> > but requested type was class Eigen::Matrix<double,3,1,0,3,1>
> ```

I have the exact problem when I compile the tests on windows, and fix it on my PR.
@etamarlu Can you try my PR if it work for you?