borglab/gtsam #1697: sudo make install gives cxx11 error

tanmayd26, 2023-12-22:
OS: Ubuntu 18.04
cmake : catkin_make -DCMAKE_C_COMPILER=gcc-6 -DCMAKE_CXX_COMPILER=g++-6
```
../gtsam/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_split_cmpts()'
../gtsam/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::status(std::experimental::filesystem::v1::__cxx11::path const&)'
../gtsam/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::_M_find_extension() const'
../gtsam/libgtsam.so.4.3a0: undefined reference to `std::experimental::filesystem::v1::__cxx11::path::parent_path() const'
collect2: error: ld returned 1 exit status
examples/CMakeFiles/UGM_chain.dir/build.make:105: recipe for target 'examples/UGM_chain' failed
make[2]: *** [examples/UGM_chain] Error 1
CMakeFiles/Makefile2:26580: recipe for target 'examples/CMakeFiles/UGM_chain.dir/all' failed
make[1]: *** [examples/CMakeFiles/UGM_chain.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

```

ProfFan, 2023-12-26:
This is most likely not a GTSAM issue, is this from another project?

tanmayd26, 2023-12-26:
So I am trying to build GTSAM on my PC. I am eventually going to use it for another project.

tanmayd26, 2023-12-27:
`-- GTSAM is a shared library due to GTSAM_FORCE_SHARED_LIB
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Boost 1.65 found.
-- Found Boost components:
   serialization;system;filesystem;thread;program_options;date_time;timer;chrono;regex
-- Found Eigen version: 3.4.0
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
CMake Warning at cmake/HandleTBB.cmake:4 (find_package):
  By not providing "FindTBB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TBB", but
  CMake did not find one.

  Could not find a package configuration file provided by "TBB" (requested
  version 4.4) with any of the following names:

    TBBConfig.cmake
    tbb-config.cmake

  Add the installation prefix of "TBB" to CMAKE_PREFIX_PATH or set "TBB_DIR"
  to a directory containing one of the above files.  If "TBB" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:79 (include)

-- Building 3rdparty
-- Found GeographicLib: /usr/lib/x86_64-linux-gnu  
CMake Warning at gtsam/3rdparty/CMakeLists.txt:80 (message):
  GeographicLib is installed on your system and GTSAM_INSTALL_GEOGRAPHICLIB
  is enabled.  Installing gtsam will either overwrite the installed
  GeographicLib or install a second version that may conflict.  You may want
  to disable GTSAM_INSTALL_GEOGRAPHICLIB if the installed version was not
  installed by GTSAM.

-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3a0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3a0
-- Install prefix: /usr/local
-- Wrote /home/autorally/installations/gtsam/build/GTSAMConfig.cmake
-- Wrote /home/autorally/installations/gtsam/build/GTSAM_UNSTABLEConfig.cmake
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 7.5.0
--  CMake version                                    : 3.10.2
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/make
-- Build flags                                               
[... 32 lines cut ...]
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Build with Memory Sanitizer                      : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Enable branch merging in DecisionTree            : Enabled
--  Allow features deprecated in GTSAM 4.3           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
CMake Warning at cmake/HandleFinalChecks.cmake:3 (message):
  TBB 4.4 or newer was not found - this is ok, but note that GTSAM
  parallelization will be disabled.  Set GTSAM_WITH_TBB to 'Off' to avoid
  this warning.
Call Stack (most recent call first):
  CMakeLists.txt:159 (include)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/autorally/installations/gtsam/build
`

varunagrawal, 2023-12-29:
Check you GNU compiler version. For the VirtualBox, it seems like GNU 7.5.0 is being picked up, maybe another one is being picked up on your Desktop that doesn't have the minimum support requirements.

tanmayd26, 2024-01-02:
Yes it is 
`gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
`

varunagrawal, 2024-01-05:
Your CMake output is only showing warnings, which is fine and doesn't affect the compilation process. The original error you posted seems to be the result of running `make`, but you are pasting the output of `cmake`. Are you sure you're compiling and installing GTSAM correctly? This could be a configuration issue.

Either way, it is recommended you ask this on the Google Group (as specified in the issue template) since it doesn't seem to be a code issue.

tanmayd26, 2024-01-05:
Yes sorry the issue seems to be with make and not cmake. Do you think if I use conda environment this issue will be solved?

varunagrawal, 2024-01-14:
I don't know if conda would help since it is more for Python, but it wouldn't hurt to try.

talregev, 2024-01-15:
@tanmayd26 
You can try this:
https://stackoverflow.com/questions/33149878/experimentalfilesystem-linker-error
Maybe it cmake problem for old gcc.

varunagrawal, 2024-02-09:
I vote to close this issue since it seems to be a configuration issue on @tanmayd26's PC. Unfortunately we cannot help in such cases since we don't have access to your machine or an identical resource to debug on.

I highly recommend getting someone in your lab with experience configuring software to take a look at this. It may just be a path issue where the older GCC is getting picked over the newer GCC. Sometimes installing a newer version of GCC and deleting the build folder to do a full rebuild can work.

closed 2024-02-09 by varunagrawal