borglab/gtsam #1699: Jacobian calculate in Pose3 translation function is correct?

Little99lantern, 2023-12-29:
the file is geometry/Pose3.cpp
according to the code. I think Lie Group with Vectot6 is omega(3 dimensions) and translation(3 dimensions) 
sou why the jacobian is not  Z_3x3, I_3x3 ??
is there anything wrong with me?
need help thanks
```
/* ************************************************************************* */
const Point3& Pose3::translation(OptionalJacobian<3, 6> Hself) const {
  if (Hself) *Hself << Z_3x3, rotation().matrix();
  return t_;
}
```

varunagrawal, 2023-12-29:
The `translation` function is defined as $f: R^6 \rightarrow R^3$, which is why the Jacobian has to be 3x6.

closed 2023-12-29 by varunagrawal

Little99lantern, 2023-12-30:
thanks for your attention. 
yep, I know that is 3x6. I mean it is  (Z_3x3, I_3x3) but not (Z_3x3, rotation().matrix())

varunagrawal, 2023-12-30:
That's because translations are represented in the rotated coordinate frame for a pose $t_W = R^W_B * t_B$. A look at a textbook on robotic pose configuration (e.g. Murray & Sastry) should help with this understanding.

Little99lantern, 2023-12-31:
thank you very much. I got it.
By the way, happy new year.