borglab/gtsam #1701: Will the new GTSAM 4.3 be binary compatible with earlier versions of GTSAM 4.x?

berndpfrommer, 2023-12-30:
Quoting from the README.md file:

> As of January 2023, the develop branch is officially in "Pre 4.3" mode. We envision several API-breaking changes as we switch to C++17 and away from boost.

Will this break binary compatibility? Just asking because doing so would require bumping the major version number to avoid packaging mayhem.

ProfFan, 2024-01-02:
Are you referring to the API? I believe deprecated APIs should still be available, ABI-wise I don't know.

berndpfrommer, 2024-01-08:
I was referring to ABI. If we break ABI then we'll have to bump the major version number at the package level. Here is a tool that apparently can be used to check if the ABI changes between revisions.
https://lvc.github.io/abi-compliance-checker/
At the moment GTSAM is not officially distributed with Ubuntu but through a personal package archive PPA. So we'll simply ask people to uninstall the old version and install the new version of GTSAM. Once GTSAM becomes an official Debian/Ubuntu package, this will no longer be possible: gtsam 4.3 would have to be binary compatible with gtsam 4.2.
Just wanted to bring awareness to this aspect. I will let this issue simmer for a little longer and then close it.

berndpfrommer, 2024-01-12:
OK, closing this issue since it is not on the radar screen until GTSAM will be released as a proper Ubuntu package.

closed 2024-01-12 by berndpfrommer