borglab/gtsam #1702: Included cephes code should contain a provenance and license

dkogan, 2024-01-01:
Hi. I see that another external library was added to the gtsam sources: https://github.com/borglab/gtsam/tree/develop/gtsam/3rdparty/cephes

Looking around, I should clearly be able two things, that I cannot find:

- Where this came from
- The distribution license

Do we know these? Are you allowed to use this code? Under what conditions?

Thanks

ProfFan, 2024-01-02:
@varunagrawal

varunagrawal, 2024-01-02:
Cephes is hosted as source on netlib.org. 

https://github.com/borglab/gtsam/pull/1572
https://www.netlib.org/cephes/

Unfortunately there is no license specified with the source but when I looked at the author's project SMath, there is a page which specifies it comes with a MIT license which allows redistribution without conditions.

https://smath.com/en-US/view/CephesMathLibrary/license

I didn't add a license or source because it wasn't already present and I wanted to make minimal changes (particularly because I'm not a lawyer).

varunagrawal, 2024-01-02:
I'll make a PR to update the Cephes readme and add the license in a bit.

dkogan, 2024-01-02:
Hi. This feels very insufficient. The MIT-license notice at the smath.com site applies to whatever is distributed at the smath.com site, which is binaries built for Windows, not sources. There's a link to an svn repo, but it's dead. Did the copy you're including in your git tree come from smath.com or from netlib? If from netlib, then you need a license from netlib.

netlib DOES have a readme: https://www.netlib.org/cephes/readme

That has no clear statement either, but it strongly suggests that you CANNOT use this without any extra explicit license.

Do you really need this? There are no public implementations of whatever you're using here?

varunagrawal, 2024-01-02:
The readme states "What you see here may be used freely but it comes with no support or guarantee." which I took to mean that we can vendor the source as required.

Moreover, the SMath site states that Cephes is open source under the MIT license and is hosted by the authors of Cephes. I'm not sure where that means that it applies to only code/source from the SMath site only. Sufficiency can be debated in this case.

We need an implementation of the inverse incomplete gamma function which used to be provided by Boost but we've moved away from Boost, hence the need for an alternative. Cephes is well tested (used by scipy), is open source, provides a liberal MIT license, and has a very small memory footprint, which made it a stellar choice for our needs.

dkogan, 2024-01-02:
Hi.

Varun Agrawal ***@***.***> writes:

> The readme states "What you see here may be used freely but it comes
> with no support or guarantee." which I took to mean that we can vendor
> the source as required.

The text says

  Some software in this archive may be from the book _Methods and
  Programs for Mathematical Functions_ (Prentice-Hall or Simon &
  Schuster International, 1989) or from the Cephes Mathematical Library,
  a commercial product. In either event, it is copyrighted by the
  author. What you see here may be used freely but it comes with no
  support or guarantee.

I missed the "In either event, it is copyrighted by the author" part.
That makes it OK in my mind. Thanks for making me look again.

And since you mentioned scipy, I looked at its state in Debian just now,
and cephes is included in its sources as you say, with its own copyright
statement:

  https://sources.debian.org/src/scipy/1.10.1-6/debian/copyright/#L181

It probably shouldn't say "bsd-3-clause", but no full statement exists
as you saw, so maybe that's good-enough.

Thanks much for discussing.

talregev, 2024-01-03:
The svn repo is working:
https://smath.info:8443/!/#public/view/head/plugins/CephesMathLibrary

dkogan, 2024-01-03:
Hi. The link you just posted does work, but the link from the page linked earlier in this post is broken. So if I go here: https://smath.com/en-US/view/CephesMathLibrary/license and click on the "SVN repository" link on that page, I get

```
Error loading stylesheet: A network error occurred loading an XSLT stylesheet:
https://smath.info/!/assets/xsl/svnindex-ng.xsl?v=2
```

talregev, 2024-01-03:
You can read the readme here:
https://smath.info:8443/!/#public/view/head/plugins/CephesMathLibrary/mathlib/ReadMe.txt

closed 2024-01-12 by varunagrawal