borglab/gtsam #1708: Factor between two landmarks

pie-leroy, 2024-01-09:
Hello, I am looking to develop a SLAM using the gtsam Python API. In my project, I am doing image segmentation which allows me to extract a landmark for each object of interest. I can make assumptions about the distances between two landmarks, i.e., a minimum distance. My landmarks are 3D Points.

I feel like I could use "ReferenceFrameFactor" but I haven't succeeded in Python.
Is it possible to insert a factor between two landmarks? If yes, how?
Thank you in advance.

ProfFan, 2024-01-10:
Hi, is BetweenFactor good for this? Is there anything special?

pie-leroy, 2024-01-15:
hello, i can't import BetweenFactor in python

pie-leroy, 2024-01-15:
Thank you very much, indeed it works, here is the code used

`
noise_model = gtsam.noiseModel.Isotropic.Sigma(3, 0.0001) 
distance_factor = gtsam.BetweenFactorPoint3(symbol('l', 0), symbol('l', 24), np.array([0.0, 0.0,  553]), noise_model)
graph.add(distance_factor)
result = optimize(graph, initial_estimate)`

closed 2024-01-15 by pie-leroy