borglab/gtsam #1724: ImportError in case of custom build of python package

Mikhail2017, 2024-02-24:
Hi. I'm building a python package according to the instruction
https://github.com/borglab/gtsam/blob/develop/python/README.md
, but when I import a package like
`from gtsam import gtsam, utils`
it produces error

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mika/.local/lib/python3.11/site-packages/gtsam/__init__.py", line 9, in <module>
    from gtsam import gtsam, utils
ImportError: /home/mika/.local/lib/python3.11/site-packages/gtsam/gtsam.cpython-311-x86_64-linux-gnu.so: undefined symbol: _ZTIN5gtsam11TableFactorE
```

Similar error is triggered in case of earlier versions too.
Strange, but I didn't face this issue in case of installing prebuilt package using pip.

ProfFan, 2024-02-25:
Are you absolutely sure that you do not have a second copy of GTSAM libraries installed? Sees like one

Mikhail2017, 2024-02-25:
Thanks. It was a reason.

closed 2024-02-25 by Mikhail2017