borglab/gtsam #1727: Ubuntu package builds failing

berndpfrommer, 2024-02-29:
## Description

## Steps to reproduce

1. 
2.

## Expected behavior

## Environment

## Additional information
Got this error log from the automated Ubuntu builds. Likely caused by one of the last few commits.
```
<!-- Add any other infor or context about the problem here. -->
<<PKGBUILDDIR>>/gtsam/geometry/Cal3Bundler.cpp
In file included from /<<PKGBUILDDIR>>/gtsam/geometry/Point2.h:21,
                 from /<<PKGBUILDDIR>>/gtsam/geometry/Cal3.h:24,
                 from /<<PKGBUILDDIR>>/gtsam/geometry/Cal3.cpp:18:
/<<PKGBUILDDIR>>/gtsam/base/std_optional_serialization.h:13: error: unterminated #ifdef
   13 | #ifdef GTSAM_ENABLE_BOOST_SERIALIZATION
      | 
In file included from /<<PKGBUILDDIR>>/gtsam/geometry/Point2.h:21,
                 from /<<PKGBUILDDIR>>/gtsam/geometry/Cal3.h:24,
                 from /<<PKGBUILDDIR>>/gtsam/geometry/Cal3Bundler.h:22,
                 from /<<PKGBUILDDIR>>/gtsam/geometry/Cal3Bundler.cpp:20:
/<<PKGBUILDDIR>>/gtsam/base/std_optional_serialization.h:13: error: unterminated #ifdef
   13 | #ifdef GTSAM_ENABLE_BOOST_SERIALIZATION
      | 
make[3]: *** [gtsam/CMakeFiles/gtsam.dir/build.make:289: gtsam/CMakeFiles/gtsam.dir/geometry/Cal3.cpp.o] Error 1
```

berndpfrommer, 2024-02-29:
The offending file [std_optional_serialization.h](
https://github.com/borglab/gtsam/blob/01490055ffbe6ee0f5097d8c6325f54d14c3d72d/gtsam/base/std_optional_serialization.h) has 5 lines starting with ``#if`` but only 4 lines starting with ``#endif``.

arutkowski, 2024-03-06:
I have also encountered this issue recently. The simple fix seems to be to add an additional `#endif` at the end of the header file, but I didn't spend a lot of time to make sure that it is the correct fix.

arutkowski, 2024-03-18:
I'm pretty sure this issue was fixed by [this commit](https://github.com/borglab/gtsam/commit/4947a101329772fbe530eb773d232ce3e5d5c744), or perhaps a prior commit.

berndpfrommer, 2024-03-20:
OK, looks like it's building now, so closing this issue.

closed 2024-03-20 by berndpfrommer