borglab/gtsam #1734: `len(CameraSetCal3Bundler)` throw RecursionError

zitongzhan, 2024-03-17:
## Description
`CameraSetCal3Bundler` Python binding report RecursionError only when compiled from source of the newest develop branch. The simplest steps to reproduce are included below. Problem did not appear when 4.2 version was installed using `pip install gtsam`
```
>>> from gtsam import CameraSetCal3Bundler
>>> len(CameraSetCal3Bundler())
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RecursionError: maximum recursion depth exceeded while calling a Python object
```

## Steps to reproduce

1. Compile from main with python binding enabled. 
```
mkdir -p build ; cd build
cmake .. -DGTSAM_PYTHON_VERSION=3.10.13 -DGTSAM_BUILD_PYTHON=1 -DGTSAM_WITH_TBB=ON
```
2. Run the above code snippet. 

## Expected behavior

## Environment

CMake output:
```
-- The CXX compiler identification is GNU 11.4.0
-- The C compiler identification is GNU 11.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM is a shared library due to GTSAM_FORCE_SHARED_LIB
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- Found Boost: /workspace/mambaforge/envs/gtsfm-v1/lib/cmake/Boost-1.84.0/BoostConfig.cmake (found suitable version "1.84.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.4.0
-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
CMake Warning at cmake/HandleTBB.cmake:4 (find_package):
  By not providing "FindTBB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TBB", but
  CMake did not find one.

  Could not find a package configuration file provided by "TBB" (requested
  version 4.4) with any of the following names:

    TBBConfig.cmake
    tbb-config.cmake

  Add the installation prefix of "TBB" to CMAKE_PREFIX_PATH or set "TBB_DIR"
  to a directory containing one of the above files.  If "TBB" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:80 (include)

-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3a0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3a0
-- Install prefix: /usr/local
-- Found Python: /workspace/mambaforge/envs/gtsfm-v1/bin/python3.10 (found suitable exact version "3.10.13") found components: Interpreter Development Development.Module Development.Embed 
-- gtwrap Package config : /usr/local/lib/cmake/gtwrap
-- gtwrap version        : 1.0
-- gtwrap CMake path     : /usr/local/lib/cmake/gtwrap
-- gtwrap library path   : /usr/local/lib/gtwrap
-- gtwrap binary path    : /usr/local/bin/gtwrap
-- gtwrap header path    : /usr/local/include/gtwrap
-- Checking Python Version
-- Setting Python version for wrapper
-- pybind11 v2.11.1 
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Wrote /workspace/gtsam/build/GTSAMConfig.cmake
-- Wrote /workspace/gtsam/build/GTSAM_UNSTABLEConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 11.4.0
--  CMake version                                    : 3.22.1
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/gmake
-- Build flags                                               
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Enabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            :  -O3 -DNDEBUG
--  Enable Boost serialization                       : ON
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_17
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  Use System Eigen                                 : OFF (Using version: 3.4.0)
--  Use System Metis                                 : OFF
--  Using Boost version                              : 1.84.0
--  Use Intel TBB                                    : TBB not found
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : STL
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : No
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Build with Memory Sanitizer                      : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Enable branch merging in DecisionTree            : Enabled
--  Allow features deprecated in GTSAM 4.3           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Enabled
--  Python version                                   : 3.10.13
-- ===============================================================
CMake Warning at cmake/HandleFinalChecks.cmake:3 (message):
  TBB 4.4 or newer was not found - this is ok, but note that GTSAM
  parallelization will be disabled.  Set GTSAM_WITH_TBB to 'Off' to avoid
  this warning.
Call Stack (most recent call first):
  CMakeLists.txt:160 (include)

-- Configuring done
-- Generating done
-- Build files have been written to: /workspace/gtsam/build
```
## Additional information

varunagrawal, 2024-03-20:
This is an issue with the way we currently wrap the CameraSet class via `bind_vector`. Creating typedefs of the `CameraSet` wrapped class (`typedef gtsam::CameraSet<gtsam::PinholeCamera<gtsam::Cal3_S2>> CameraSetCal3_S2;`) works correctly, but causes tests to fail since we don't have backwards compatibility for the constructor.

Maybe investigating why the `bind_vector` fails would help.

zitongzhan, 2024-03-21:
Since the pip package doesn't have the problem, I am wondering what is the environment (e.g. pybind, gcc, and the exact Cmake options) used to compile the wheel. I am happy to do further testings to see which version caused the problem.

zitongzhan, 2024-03-21:
> This is an issue with the way we currently wrap the CameraSet class via `bind_vector`. Creating typedefs of the `CameraSet` wrapped class (`typedef [...]

Since the pip package doesn't have the problem, I am wondering what is the environment (e.g. pybind, gcc, and the exact Cmake options) used to compile the wheel. I am happy to do further testings to see which version caused the problem.

ProfFan, 2024-03-21:
Can you give a GDB backtrace? Just attach gdb to the python process and set break on every C++ exception. Thank you very much!

varunagrawal, 2024-03-24:
I reverted my local GTSAM to the version before the 4.2 python wheels release and I am still getting the same error. I am not sure what the issue is.

zitongzhan, 2024-03-24:
> I reverted my local GTSAM to the version before the 4.2 python wheels release and I am still getting the same error. I am not sure what the issue is.

Same when reverting to the 4.2 and 4.1 tag. The 4.2 pypi package has no issue though.