borglab/gtsam #1736: Built GTSAM to local directory: error while loading libcephes-gtsam.so.1 (with solution and questions)

diplodocuslongus, 2024-03-21:
**TL,DR:** 
gtsam was built to a local user folder on  Linux. Running a simple example result in `error while loading shared libraries: libcephes-gtsam.so.1...`
The issue can be solved with:
`export LD_LIBRARY_PATH=~/libs/gtsam/lib`
But I don't think I should have to do this. Related: https://github.com/borglab/gtsam/issues/380 .

**More details:**
I've just built gtsam (current develop branch) in a local user folder on my Pop!_OS (ubuntu 22.04 / debian based) Linux box:

```
cmake -DGTSAM_BUILD_PYTHON=ON -DCMAKE_INSTALL_PREFIX=~/libs/gtsam -DGTSAM_USE_QUATERNIONS=ON  ..
make
make install
```

Now on to testing the build with a simple example, I get the following when running the executable:
`error while loading shared libraries: libcephes-gtsam.so.1: cannot open shared object file: No such file or directory`

The example is the `OdometryExample.cpp` (copied directly from the example folder).
My CMakeLists.txt:
```
project(OdometryExample)
cmake_minimum_required(VERSION 3.0)
# points to my local build target directory
set(GTSAM_DIR "~/libs/gtsam/lib/cmake/GTSAM")
find_package(GTSAM REQUIRED)

include_directories(${GTSAM_INCLUDE_DIR})
# for debug info
get_target_property(gtsam_link_lib gtsam INTERFACE_LINK_LIBRARIES)
message("${gtsam_link_lib}")

add_executable(OdometryExample OdometryExample.cpp)
target_link_libraries(OdometryExample gtsam)
```

Running `cmake` shows the following as a result of `message("${gtsam_link_lib}")`:
`Boost::serialization;Boost::system;Boost::filesystem;Boost::thread;Boost::date_time;Boost::regex;Boost::timer;Boost::chrono;TBB::tbb;TBB::tbbmalloc;metis-gtsam-if;cephes-gtsam-if;gtsam_eigen3
`

`make` doesn't complain.
Digging deeper to try to debug the issue:
All the gtsam examples were built during the gtsam installation (`make` and `make install`), and executing the Odometry example from the build directory (`my_localpath_to_gtsam/gtsam/build/examples`) doesn't complain about a missing `libcephes-gtsam.so.1`.

Content of the target local `lib` directory (libcephes-gtsam.so.1 is there all right):
```
$ pwd
~/libs/gtsam/lib
$ ls
-rw-r--r-- 1 683K Mar 21 14:55 libmetis-gtsam.a
-rw-r--r-- 1 231K Mar 21 14:55 libcephes-gtsam.so.1.0.0
-rw-r--r-- 1  34K Mar 21 14:55 libCppUnitLite.a
-rw-r--r-- 1 7.7M Mar 21 15:04 libgtsam.so.4.3a0
-rw-r--r-- 1 2.0M Mar 21 15:11 libgtsam_unstable.so.4.3a0
lrwxrwxrwx 1   24 Mar 21 15:31 libcephes-gtsam.so.1 -> libcephes-gtsam.so.1.0.0
lrwxrwxrwx 1   20 Mar 21 15:31 libcephes-gtsam.so -> libcephes-gtsam.so.1
lrwxrwxrwx 1   17 Mar 21 15:31 libgtsam.so.4 -> libgtsam.so.4.3a0
lrwxrwxrwx 1   13 Mar 21 15:31 libgtsam.so -> libgtsam.so.4
lrwxrwxrwx 1   26 Mar 21 15:31 libgtsam_unstable.so.4 -> libgtsam_unstable.so.4.3a0
lrwxrwxrwx 1   22 Mar 21 15:31 libgtsam_unstable.so -> libgtsam_unstable.so.4
drwxrwxr-x 4 4.0K Mar 21 15:31 gtwrap
drwxrwxr-x 6 4.0K Mar 21 15:31 cmake
```

 Inspecting `libgtsam.so` with `ldd`:

```
$ ldd libgtsam.so
	linux-vdso.so.1 (0x00007fff1f5d1000)
	libboost_serialization.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_serialization.so.1.74.0 (0x00007801b605c000)
	libboost_timer.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_timer.so.1.74.0 (0x00007801b59f6000)
	libtbb.so.12 => /lib/x86_64-linux-gnu/libtbb.so.12 (0x00007801b59a6000)
	libtbbmalloc.so.2 => /lib/x86_64-linux-gnu/libtbbmalloc.so.2 (0x00007801b5966000)
	libcephes-gtsam.so.1 => not found
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007801b5600000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007801b587f000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007801b585f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007801b5200000)
	/lib64/ld-linux-x86-64.so.2 (0x00007801b60c7000)
	libboost_chrono.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_chrono.so.1.74.0 (0x00007801b5854000)
```

The `libcephes-gtsam.so.1 => not found` certainly shouldn't show up!

Note: `sudo ldconfig` was irrelevant here since I installed gtsam locally. 

So: 

- Is anything missing in my CMakeLists.txt?
- Why does the example built during the gtsam installation run without complaining about a missing `libcephes-gtsam.so.1` but building the same example separately does?
- Why should I have to `export LD_LIBRARY_PATH=~/libs/gtsam/lib`?
- Since the examples built during the  gtsam installation run fine, was a `LD_LIBRARY_PATH` set during installation?

diplodocuslongus, 2024-03-22:
**Updates**.

1) after a reboot (and a good night sleep), rebuilding the Odometry example with the same cmake now gives the following error and no more the libcephese related error (!!!):

```
$ ./OdometryExample 
./OdometryExample: symbol lookup error: ./OdometryExample: undefined symbol: _ZN5gtsam19DefaultKeyFormatterB5cxx11E
```
This goes for both the example built when installing gtsam (the deploy version) and re-building the example in a separate test directory.

If this helps, `c++filt` on the undefined symbol returns:

```
c++filt _ZN5gtsam19DefaultKeyFormatterB5cxx11E
gtsam::DefaultKeyFormatter[abi:cxx11]

```

I wonder: could `cxx11` means here that the build was done with c++11 rather than c++17? Likely not since the result of my `cmake` command (used to build the develop branch yesterday) shows:
`GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_17`

2) Version 4.2 (downloaded from [here](https://github.com/borglab/gtsam/releases/tag/4.2)) and also built to a local lib user folder with the same cmake command as reported in the post above doesn't complain about the libcephese library not being found and the example executes normally.

```
$ ./OdometryExample 

Factor Graph:
size: 3

Factor 0: PriorFactor on 1
  prior mean:  (0, 0, 0)
  noise model: diagonal sigmas [0.3; 0.3; 0.1];

Factor 1: BetweenFactor(1,2)
  measured:  (2, 0, 0)
  noise model: diagonal sigmas [0.2; 0.2; 0.1];

Factor 2: BetweenFactor(2,3)
  measured:  (2, 0, 0)
  noise model: diagonal sigmas [0.2; 0.2; 0.1];

Initial Estimate:
...

```
Well, checking the content of `path_to_gtsam_lib/lib` reveals that `libcephes` isn't there at all, cephes seems to have been introduced after 4.2 then.

varunagrawal, 2024-04-03:
`gtsam::DefaultKeyFormatter[abi:cxx11]` indicates the ABI is for C++11, so yes it is being compiled for C++11. Is it possible you have both a C++11 version and a C++17 version installed and your system path configuration is causing the C++11 version to be picked up?
This is especially possible if you installed to root aka `sudo make install` or installed gtsam via `apt` in the past. This would explain why the issue goes away when you export the `LD_LIBRARY_PATH`.

varunagrawal, 2024-04-22:
@diplodocuslongus any updates on this? Did you manage to resolve it?

diplodocuslongus, 2024-04-22:
Varun, I hadn't installed GTSAM before (be it via apt or build) and no, my installation was (still is) to a local folder, no `sudo`. 
I haven't dig more into the issue as I could manage with an export LD_LIBRARY_PATH. 

Not sure if this should be closed or not, I'll try to find some time to test on a brand new linux box.

varunagrawal, 2024-07-12:
I am closing this for now. Please re-open if the issue still persists.

closed 2024-07-12 by varunagrawal