borglab/gtsam #1738: Build failure for tests with GCC 13

ottojo, 2024-03-22:
## Description

`testStdOptionalSerialization.cpp` fails to build with `error: invalid use of incomplete type ‘struct boost::serialization::U’`.

## Steps to reproduce

```console
foo@bar:~$ g++ --version
g++ (GCC) 13.2.1 20230801
```

```
mkdir build && cd build
cmake ..
make check
```
Build failure:
```
[ 40%] Building CXX object gtsam/base/tests/CMakeFiles/testStdOptionalSerialization.dir/testStdOptionalSerialization.cpp.o
In file included from /usr/include/c++/13.2.1/bits/stl_pair.h:60,
                 from /usr/include/c++/13.2.1/bits/stl_algobase.h:64,
                 from /usr/include/c++/13.2.1/bits/specfun.h:43,
                 from /usr/include/c++/13.2.1/cmath:3699,
                 from /home/jonas/Workspace/gtsam/CppUnitLite/Test.h:25,
                 from /home/jonas/Workspace/gtsam/CppUnitLite/TestHarness.h:23,
                 from /home/jonas/Workspace/gtsam/gtsam/base/tests/testStdOptionalSerialization.cpp:1:
/usr/include/c++/13.2.1/type_traits: In instantiation of ‘constexpr const bool std::is_trivially_copy_constructible_v<boost::serialization::U>’:
/usr/include/boost/serialization/split_free.hpp:58:13:   required from ‘static void boost::serialization::free_loader<Archive, T>::invoke(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = std::optional<int>]’
/usr/include/boost/serialization/split_free.hpp:74:18:   required from ‘void boost::serialization::split_free(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = std::optional<int>]’
/home/jonas/Workspace/gtsam/gtsam/base/std_optional_serialization.h:103:35:   required from ‘void boost::serialization::serialize(Archive&, std::optional<_Up>&, unsigned int) [with Archive = boost::archive::text_iarchive; T = int]’
/usr/include/boost/serialization/serialization.hpp:109:14:   required from ‘void boost::serialization::serialize_adl(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = std::optional<int>]’
/usr/include/boost/archive/detail/iserializer.hpp:189:40:   [ skipping 3 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/boost/archive/detail/iserializer.hpp:463:22:   required from ‘static void boost::archive::detail::load_non_pointer_type<Archive>::invoke(Archive&, T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/detail/iserializer.hpp:626:18:   required from ‘void boost::archive::load(Archive&, T&) [with Archive = text_iarchive; T = std::optional<int>]’
/usr/include/boost/archive/detail/common_iarchive.hpp:67:22:   required from ‘void boost::archive::detail::common_iarchive<Archive>::load_override(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/basic_text_iarchive.hpp:70:52:   required from ‘void boost::archive::basic_text_iarchive<Archive>::load_override(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/text_iarchive.hpp:82:52:   required from ‘void boost::archive::text_iarchive_impl<Archive>::load_override(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/detail/interface_iarchive.hpp:68:36:   required from ‘Archive& boost::archive::detail::interface_iarchive<Archive>::operator>>(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/home/jonas/Workspace/gtsam/gtsam/base/tests/testStdOptionalSerialization.cpp:26:9:   required from here
/usr/include/c++/13.2.1/type_traits:3279:7: error: invalid use of incomplete type ‘struct boost::serialization::U’ [-fpermissive]
 3279 |     = __is_trivially_constructible(_Tp, __add_lval_ref_t<const _Tp>);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/boost/serialization/shared_ptr.hpp:29,
                 from /home/jonas/Workspace/gtsam/gtsam/base/serialization.h:31,
                 from /home/jonas/Workspace/gtsam/gtsam/base/serializationTestHelpers.h:27,
                 from /home/jonas/Workspace/gtsam/gtsam/base/tests/testStdOptionalSerialization.cpp:8:
/usr/include/boost/serialization/shared_ptr_helper.hpp:45:16: note: forward declaration of ‘struct boost::serialization::U’
   45 |     SPT< class U > &t,
      |                ^
/usr/include/c++/13.2.1/type_traits: In instantiation of ‘constexpr const bool std::is_trivially_move_constructible_v<boost::serialization::U>’:
/usr/include/boost/serialization/split_free.hpp:58:13:   required from ‘static void boost::serialization::free_loader<Archive, T>::invoke(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = std::optional<int>]’
/usr/include/boost/serialization/split_free.hpp:74:18:   required from ‘void boost::serialization::split_free(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = std::optional<int>]’
/home/jonas/Workspace/gtsam/gtsam/base/std_optional_serialization.h:103:35:   required from ‘void boost::serialization::serialize(Archive&, std::optional<_Up>&, unsigned int) [with Archive = boost::archive::text_iarchive; T = int]’
/usr/include/boost/serialization/serialization.hpp:109:14:   required from ‘void boost::serialization::serialize_adl(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = std::optional<int>]’
/usr/include/boost/archive/detail/iserializer.hpp:189:40:   [ skipping 3 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/boost/archive/detail/iserializer.hpp:463:22:   required from ‘static void boost::archive::detail::load_non_pointer_type<Archive>::invoke(Archive&, T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/detail/iserializer.hpp:626:18:   required from ‘void boost::archive::load(Archive&, T&) [with Archive = text_iarchive; T = std::optional<int>]’
/usr/include/boost/archive/detail/common_iarchive.hpp:67:22:   required from ‘void boost::archive::detail::common_iarchive<Archive>::load_override(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/basic_text_iarchive.hpp:70:52:   required from ‘void boost::archive::basic_text_iarchive<Archive>::load_override(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/text_iarchive.hpp:82:52:   required from ‘void boost::archive::text_iarchive_impl<Archive>::load_override(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/detail/interface_iarchive.hpp:68:36:   required from ‘Archive& boost::archive::detail::interface_iarchive<Archive>::operator>>(T&) [with T = std::optional<int>; Archive = boost::archive::text_iarchive]’
/home/jonas/Workspace/gtsam/gtsam/base/tests/testStdOptionalSerialization.cpp:26:9:   required from here
/usr/include/c++/13.2.1/type_traits:3282:7: error: invalid use of incomplete type ‘struct boost::serialization::U’ [-fpermissive]
 3282 |     = __is_trivially_constructible(_Tp, __add_rval_ref_t<_Tp>);
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/serialization/shared_ptr_helper.hpp:45:16: note: forward declaration of ‘struct boost::serialization::U’
   45 |     SPT< class U > &t,
      |                ^
make[3]: *** [gtsam/base/tests/CMakeFiles/testStdOptionalSerialization.dir/build.make:76: gtsam/base/tests/CMakeFiles/testStdOptionalSerialization.dir/testStdOptionalSerialization.cpp.o] Error 1
make[2]: *** [CMakeFiles/Makefile2:4347: gtsam/base/tests/CMakeFiles/testStdOptionalSerialization.dir/all] Error 2
[... 128 lines cut ...]
--  Enable branch merging in DecisionTree            : Enabled
--  Allow features deprecated in GTSAM 4.3           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
-- Configuring done (4.5s)
-- Generating done (2.8s)
-- Build files have been written to: /home/jonas/Workspace/gtsam/build_13
```
</details>

## Expected behavior

Successful build

## Environment

* Kernel `6.7.10-1-MANJARO`
* Git commit 11409b03ab9b5cca450176c478ddc1190cd8f93a
* GCC `13.2.1 20230801`
* Boost `1.83`

varunagrawal, 2024-03-24:
Seems like GCC fixed a bug it had because we had a conditional compilation step in `gtsam/base/serialization.h`. The fix should be simple: check if the GCC version is `>=7` and `<13` for that block. Can you make a PR? I can help land it.

ottojo, 2024-03-24:
I think this relates to the workaround in https://github.com/borglab/gtsam/blob/develop/gtsam/base/std_optional_serialization.h#L30 rather than `serialization.h`, since it seems to correlate with including `<boost/serialization/shared_ptr.hpp>`, as mentioned in the comment there.

From what i can tell, GCC still exhibits the bug described in the comment (trying to instantiate objects of type `T<boost::serialization::U>` when `boost::serialization::U` doesn't exist).

But GCC<13 accepted the incomplete type U in the specializations of the form
```
namespace boost { namespace serialization { struct U; } }
namespace std { template<> struct is_trivially_default_constructible<boost::serialization::U> : std::false_type {}; }
```
which GCC 13 does not anymore.
It seems gcc rejects using incomplete types here since version 13, and clang since version 16, so perhaps it was erroneously accepted before.

The issue seems to disappear if `U` is fully declared
```diff
- namespace boost { namespace serialization { struct U; } }
+ namespace boost { namespace serialization { struct U{}; } }
```
but I'm unsure if this has any unintended side effects.

Interestingly, both GCC 12.3 and 13.2.1 don't actually seem to require the type traits, only the declaration/definition of `U`, but i haven't tested any older GCC versions...

varunagrawal, 2024-03-25:
That's a great catch and analysis. I think you've found the core of the problem and described it very nicely (with a potential fix).
I agree that it is uncertain what a full declaration of `U` will end up doing. I will need to look into this extensively, but I unfortunately don't have the time this week. I can get back to this after the 29th of March if that works?

ottojo, 2024-03-25:
No rush at all, thank you for looking into the issue! I'm not depending on using GCC13 anytime soon, it just happened to be the default on my system.

OznOg, 2024-11-10:
this was also reported by eigen in https://gitlab.com/libeigen/eigen/-/issues/1676 and is a boost issue https://github.com/boostorg/serialization/pull/144 that was fixed in later boost version.

closed 2025-01-20 by varunagrawal

Pilarinen, 2025-05-12:
> I think this relates to the workaround in https://github.com/borglab/gtsam/blob/develop/gtsam/base/std_optional_serialization.h#L30 rather than `serialization.h`, since it seems to correlate with including `<boost/serialization/shared_ptr.hpp>`, as mentioned in the comment there.
> 
> From what i can tell, GCC still exhibits the bug described in the comment (trying to instantiate objects of type `T<boost::serialization::U>` when `boost::serialization::U` doesn't exist).
> 
> But GCC<13 accepted the incomplete type U in the specializations of the form
> 
> ```
> namespace boost { namespace serialization { struct U; } }
> namespace std { template<> struct is_trivially_default_constructible<boost::serialization::U> : std::false_type {}; }
> ```
> 
> which GCC 13 does not anymore. It seems gcc rejects using incomplete types here since version 13, and clang since version 16, so perhaps it was erroneously accepted before.
> 
> The issue seems to disappear if `U` is fully declared
> 
> - namespace boost { namespace serialization { struct U; } }
> + namespace boost { namespace serialization { struct U{}; } }
> but I'm unsure if this has any unintended side effects.
> 
> Interestingly, both GCC 12.3 and 13.2.1 don't actually seem to require the type traits, only the declaration/definition of `U`, but i haven't tested any older GCC versions...

Thank you for the solution. I had the same problem, but mine was in gtsam-4.3a/gtsam/base/std_optional_serialization.h. I can't believe that I have stuck for two days just for a curly bracket {}    :(

PR #1984 (2025-01-20): Fixes
Fixes #1738 and #1982.

Additionally, replaced `[=]`, which captures by copy in lambda expressions, by `[&]` which captures by reference.

I updated the CI to check for lower and upper bounds of versions, with the assumption that everything in between is supported. Additionally, if the `no_boost` flag is set, I updated the CI to **not** install Boost so that we can better capture issues with Boost being optional.

Update: Also fixed a bug from #1444 where while the `$GITHUB_ENV` variable is updated for the No-Boost case, it isn't actually used anywhere.

cc: @Gold856