borglab/gtsam #1746: terminate called after throwing an instance of 'gtsam::ValuesKeyDoesNotExist'

JACKLiuDay, 2024-04-17:
Hi, thank you for your great work about gtsam. I tried to use FixedLagSmootherExample.cpp for my own project. But it shows error after I run my ros bag.  My gtsam is 4.0.3.
[image]
I set the lag to be 2s. But when the factor size runs to 2.x s, the program dies.

JACKLiuDay, 2024-04-18:
I just found something wrong with my code. If I use latestState = smootherBatch->calculateEstimate<Pose3>(currentKey);, the program runs without error. If I use latestState = smootherBatch->calculateEstimate(); there will be error gtsam::ValuesKeyDoseNotExist.

zzodo, 2024-04-22:
Hi,
Did you solved this issue?

I just wonder maybe you made some mistakes using the keys and symbols. (Sorry if it isn't) Other users can help you only if you provide the minimal reproducing code examples. :D

My guess is that you are using the variable `currentKey` as a `gtsam::Key`, not a symbol.
If then, try with `X(currentKey)` with the symbol shorthand.

closed 2024-04-23 by JACKLiuDay