borglab/gtsam #1751: Cannot find the Jacobian of Pose3

ZBX2333, 2024-05-08:
When I use GTSAM, build the factors that optimize for Pose3. Taking the variant of GPSFactor as an example, the residual is as follows.
`residual = R * p.translation() – nT_;`
I'm going to change GPSFactor to NoiseModelFactor2, and I'm going to be evaluating Pose3 and the coefficient R, and I should have two Jacobians, but I don't find the Jacobian for Pose3. I only found Jacobian for Pose2 in the [website](https://gtsam.org/tutorials/intro.html).
I guess the Jacobian for Pose3 would look something like this:
`Jacobian << Z_3x3 , R * p.rotation().matrix();`
But running my code produced the wrong result.

ProfFan, 2024-05-30:
You can just use Expressions (automatic differentiation), just in case you don't know

varunagrawal, 2024-09-06:
You want the Jacobian of the residual wrt the Pose3. You can derive that mathematically using pen-and-paper and plug the corresponding components in. You don't necessarily need the Jacobian of Pose3 (moreover the Jacobian of Pose3 with itself is just identity).

closed 2024-09-06 by varunagrawal