borglab/gtsam #1756: Add ExpressionFactor to Python Wrapper.

Luke20000429, 2024-05-28:
## Feature

Add ExpressionFactor to Python Wrapper so that users don't have to manually calculate the derivatives of custom factors.

## Motivation

I am building some custom factors with gtsam Python. Looks like I cannot call `gtsam.ExpressionFactor` due to "no Attribute" error. Because there are quite a lot of nonlinear constraints. It will make my work much easier if AD can be included in gtsam python.

## Alternatives

Any other way to use AD in gtsam python when building custom factors?

ProfFan, 2024-05-30:
CustomFactor are really just normal Python code so you can use any Python-based AD code to write stuff. However that also means GTSAM types cannot be used.

We have previously considered wrapping ExpressionFactors but no one in the team has any bandwidth to do it. I did a preliminary investigation on doing it, and I think it's doable, so if you want to give it a try, please do :)

Luke20000429, 2024-05-30:
Will that be easier if I just wrap the `NumericDerivative` to Python?

ProfFan, 2024-06-11:
@Luke20000429 Actually you are probably right, if numerical derivative works for you that is the easiest!

closed 2024-07-12 by varunagrawal