borglab/gtsam #1759: ROS2 Jazzy binaries not using GTSAM_USE_SYSTEM_EIGEN=ON

matlabbe, 2024-06-02:
## Description

Currently with `ros-jazzy-gtsam` package, if your code is also including Eigen from another package like PCL, there will be a compilation error when including gtsam headers like this:

```
/opt/ros/jazzy/include/gtsam/base/Vector.h:76:30: error: static assertion failed: Error: GTSAM was built against a different version of Eigen
   76 |     GTSAM_EIGEN_VERSION_MAJOR==EIGEN_MAJOR_VERSION,
      |                              ^
/opt/ros/jazzy/include/gtsam/base/Vector.h:76:30: note: the comparison reduces to '(3 == 4)'
```
GTSAM is built with internal eigen version 3.3.7, while the system eigen is 3.4.0 on Ubuntu 24.04.

It seems that this patch setting `GTSAM_USE_SYSTEM_EIGEN=ON ` hasn't been applied to jazzy release branch: https://github.com/ros2-gbp/gtsam-release/commit/0158c16db756e2ccc4720a85877d6f966f40ebc9, which is probably causing the current issue.

ProfFan, 2024-06-03:
Oh this is bad, pinging @jlblancoc for ROS2 stuff

jlblancoc, 2024-06-05:
This should be fixed after https://github.com/ros/rosdistro/pull/41495 is merged, and Jazzy goes over a release cycle. 

You could leave this one open until it's verified that downstream packages are fixed.

matlabbe, 2024-06-11:
Downstream looks good now: https://build.ros2.org/job/Jbin_uN64__gtsam__ubuntu_noble_amd64__binary/

Thanks!

matlabbe, 2024-07-23:
I am sorry to re-open this issue, but the same fix would need to be applied also to rhel9 release on ros buildfarm. Downstream packages failing: https://build.ros2.org/job/Jbin_rhel_el964__gtsam__rhel_9_x86_64__binary/

FYI @marcoag

jlblancoc, 2024-07-24:
@matlabbe I pushed a patch for RPM too: https://github.com/ros2-gbp/gtsam-release/commit/75ffb3dfcacb8f8b079f1bb965f7bb4366b01438 
Please, keep an eye on the [rosdistro PR](https://github.com/ros/rosdistro/pull/42200) and check if this change actually propagates to the build farm for rhel9.

matlabbe, 2024-07-24:
Looks good now, thank you!

closed 2024-07-24 by matlabbe