borglab/gtsam #1761: Directed factors

Luke20000429, 2024-06-05:
## Feature

Factor with two keys but only one of the key's gradients is calculated for updating, another one's gradient is kept as 0, i.e. the value of another key is not affected by this factor. 
For example, I have a landmark L1 and a robot R1, and I build a factor 
`f = directedRange2D(R1, L1, _range)`, I want `f` to only affect the value of R1.

## Motivation

In some cases, I want some robots to localize the `L1`, and others only consider it as a fixed point. However, I cannot find any feature in GTSAM that can construct a directed edge in the factor graph. 

## Pitch

As I said, suppose the original Jacobian matrix of `f` is `[dr/dR1 dr/dL1]`. For this case, let's only keep `f = [dr/dR1, 0]`

dellaert, 2024-06-05:
It's very useful to freeze some variables, I agree, and a directed graph idea is interesting.

However, I don't think we want to break the invariant that a factor graph is an undirected, bipartite graph. 

Two ways to do this currently in GTSAM are:
- by making the variable you don't want to optimize over a parameter in the factor. Maybe a bit cumbersome, I agree.
- If you are using expressions it is easy: if optimizing, construct expression with a key; if not, construct it with a constant value.

Please close this feature request if the above resolves your use case.

dellaert, 2024-06-05:
Oops, closed by accident. Re-open if needed.

closed 2024-06-05 by dellaert

Luke20000429, 2024-06-06:
I am afraid the second option is not what I want, can you elaborate on the first option? Which parameter of factor you are pointing to? Do you mean creating a custom factor?