borglab/gtsam #1763: GeographicLib not found Cmake fix

tdallapiazza, 2024-06-10:
## Description

Hi all,
First off many thanks and congratulation for your excellent work. I've been using gtsam for many years and just recently started to move my codes to build on Ubuntu 24.04LTS.
As before, cmake is not able to find geographiclib. Before, I did  symlink to the cmake modules folder of the FindGeographicLib.cmake file located at (in Ubuntu 24.04) /usr/share/cmake/geographiclib/

Facing the GeographicLib not found during configuration, I found out that an other FindGeographicLib.cmake is included in the gtsam 3rdparty CMakeLists file (this file is actually packed within gtsam code base...).

## Steps to reproduce

1. install debian package libgeographiclib-dev (the named changed in 24.04 before it was libgeographic-dev)
2. configure gtsam from a clear build directory with cmake ..
3. the ouput shows a GeographicLib not found warning

## Expected behavior

The CMakeLists.txt of the 3rdparty folder should point the FindGeographicLib.cmake file shipped by the package maintainers (even if the location is not standard).

## Environment

Fresh Ubuntu 24.04LTS installation in WSL
Cmake version 3.28
gtsam obtained from default branch (git clone https://github.com/borglab/gtsam.git)

## Additional information

So far I just patched the CMakeLists.txt in the 3rdparty folder with (line 68):
include(/usr/share/cmake/geographiclib/FindGeographicLib.cmake)

This is working on my distribution and I know this file was at the same location in Ubuntu 22.04LTS too.

I have no clue about other distributions so I can't provide a more "distribution agnostic" option...

Kindest regards,

            Thomas

ProfFan, 2024-08-14:
We need to gradually retire vendored libraries, for now I think what you can do is just remove that file from the /cmake folder.

varunagrawal, 2024-08-19:
Thomas, thank you for your kind words and brining this issue to light.
@acxz worked on providing better 3rd party support and maybe he can help with this as well? Akash has been an excellent contributor over the years!