borglab/gtsam #1770: Building error when compiling examples with boost-1.82.0

Luke20000429, 2024-06-27:
I am building gtsam from source, everything is OK except when compiling `combinedIMUexample`, etc. It gives me
```
XXXexample....libboost_program_options.so.1.82.0: undefined reference to std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
```
I turn off examples by setting `-DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF`, and it works.
Can you fix that problem?

Luke20000429, 2024-06-27:
I build the examples with 
```
cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.9.19 -DGTSAM_WITH_TBB=OFF -DGTSAM_ENABLE_BOOST_SERIALIZATION=OFF -DGTSAM_USE_BOOST_FEATURES=OFF
```
But that is not an optimal solution.

ProfFan, 2024-06-28:
linker error is most likely caused by your local environment. Could you test if the same thing happens with a new Ubuntu 20.04 environment?

Luke20000429, 2024-07-10:
Find out the reason is due to my out-of-date stdc++6.
```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt update
sudo apt upgrade
```
Cr. ChatGPT-4o

closed 2024-07-10 by Luke20000429