borglab/gtsam #1771: Numpy 2.0 arrays cause segfaults in many GTSAM constructors

meboler-snl, 2024-06-27:
## Description

Passing a numpy 2.0 array into a GTSAM constructor will result in a segfault and a kernel crash.

## Steps to reproduce
```python
import gtsam
import numpy as np

mat = np.identity(3)
R = gtsam.Rot3(mat)

```

## Expected behavior

Successful construction of an object.

## Environment

Ubuntu 22.04
GTSAM 4.2 (installed via pypi/pip)
Numpy 2.0 (installed via pypi/pip)

## Additional information

meboler-snl, 2024-06-27:
[image]

ProfFan, 2024-06-28:
Need upgrade to latest pybind11

ProfFan, 2024-07-02:
Done in #1773, please test latest master and report back if you have other problems :)

closed 2024-07-02 by ProfFan