borglab/gtsam #1778: Optmization Failed by using Unary Factor

dolgae03, 2024-07-08:
I implemented Factor Graph using gps pseudorange information.

However, the optmizer doesn't works.

Would you like to give some information. about my factor?

> Final Result:
> 
> Values with 1 values:
> Value 0: (gtsam::Pose3)
> R: [
>         0, 0, 0;
>         0, 0, 0;
>         0, 0, 0
> ]
> t: 100 100 100

Code
```C++
class GpsFactor : public NoiseModelFactorN<Pose3> {
    double pseudorange_;
    Point3 satellite_position_;

public:
    // Shorthand for a smart pointer to a factor
    using NoiseModelFactorN<Pose3>::evaluateError;
    typedef std::shared_ptr<GpsFactor> shared_ptr;

    // Constructor
    GpsFactor(Key j, double pseudorange, const Point3& satellite_position, const SharedNoiseModel& model)
        : NoiseModelFactorN<Pose3>(model, j), pseudorange_(pseudorange), satellite_position_(satellite_position) {}

    // Destructor
    ~GpsFactor() override {}

    // Evaluate error function
    Vector evaluateError(const Pose3& user_position, OptionalMatrixType H) const override {
        // Calculate the distance between the user position and the satellite position
        double dx = user_position.x() - satellite_position_.x();
        double dy = user_position.y() - satellite_position_.y();
        double dz = user_position.z() - satellite_position_.z();
        double predicted_range = std::sqrt(dx * dx + dy * dy + dz * dz);

        // If the Jacobian matrix H is provided, calculate it
        if (H) {
            (*H) = (Matrix(1, 3) << (user_position.x() - satellite_position_.x()) / predicted_range,
                    (user_position.y() - satellite_position_.y()) / predicted_range,
                    (user_position.z() - satellite_position_.z()) / predicted_range).finished();
        }

        cout << user_position.x() << " " << user_position.y() << " " <<user_position.z() << endl; 
        cout << (Vector(1) << fabs(pseudorange_ - predicted_range)).finished() << endl;

        // Return the difference between the predicted range and the pseudorange measurement
        return (Vector(1) << fabs(pseudorange_ - predicted_range)).finished();
    }

    // Clone function
    gtsam::NonlinearFactor::shared_ptr clone() const override {
        return std::static_pointer_cast<gtsam::NonlinearFactor>(
            gtsam::NonlinearFactor::shared_ptr(new GpsFactor(*this)));
    }

};

int main() {
    // 데이터 파일 경로
    std::string pr_file = "../data/pr.csv";
    std::string sv_pos_file = "../data/sv_pos.csv";
    std::string sv_vel_file = "../data/sv_vel.csv";
    std::string time_file = "../data/time.csv";

    // CSV 파일 읽기
    std::vector<std::vector<double>> pr_data = readPseudorangeCSV(pr_file);
    std::vector<std::vector<std::vector<double>>> sv_pos_data = readSVPosAndVelCSV(sv_pos_file);
    std::vector<std::vector<std::vector<double>>> sv_vel_data = readSVPosAndVelCSV(sv_vel_file);
    std::vector<std::pair<int, double>> time_data = readGpsTimeCSV(time_file);

    checkCSVData(pr_data, sv_pos_data, sv_vel_data, time_data);

    std::cout << "Here" << std::endl;

    // 노이즈 모델 설정 (예제) 
    NonlinearFactorGraph graph;
    std::shared_ptr<GpsFactor> factor;

    std::cout << "Here2" << std::endl;
    //pr_data.size()
    const size_t max_epoch = 1;
    // GpsFactor 추가
    for (size_t epoch = 0; epoch < max_epoch; ++epoch) {
        for (size_t satellite = 0; satellite < pr_data[epoch].size(); ++satellite) {
            double pr_value = pr_data[epoch][satellite];

            // NaN 값인지 확인
            if (std::isnan(pr_value)) continue;

            // 위성 위치 가져오기
            const auto& sv_position = sv_pos_data[epoch][satellite];
            Point3 sv_loc(sv_position[0], sv_position[1], sv_position[2]);

            // 팩터 그래프에 GpsFactor 추가
            auto unaryNoise = noiseModel::Diagonal::Sigmas(Vector1(1)); 
            graph.emplace_shared<GpsFactor>(epoch, pr_value, sv_loc, unaryNoise);
        }
    }

     graph.print("\nFactor Graph:\n");  // print

    // 초기 사용자 위치 설정 (예제)
    Values initial;

    for (size_t epoch = 0; epoch < max_epoch; ++epoch){
        auto pose = gtsam::Pose3(Rot3(0,0,0,0,0,0,0,0,0), Point3(100, 100, 100));
        initial.insert(epoch, pose);
    }   

    std::cout << "Here33" << std::endl;

    LevenbergMarquardtParams params;
    params.setMaxIterations(100);  // 최대 반복 횟수 설정
    params.setRelativeErrorTol(1e-5);  // 상대 오차 허용치 설정
    params.setAbsoluteErrorTol(1e-5);  // 절대 오차 허용치 설정

    // 최적화 수행
    LevenbergMarquardtOptimizer optimizer(graph, initial, params);
    Values result = optimizer.optimize();
    result.print("Final Result:\n");

    std::cout << "Here44" << std::endl;
    Marginals marginals(graph, result);
    std::cout << "x1 covariance:\n" << marginals.marginalCovariance(0) << std::endl;

    return 0;
}

```

varunagrawal, 2024-07-12:
Your factor seems fine to me, but the failure of the optimizer could be due to the mismatch of frames in the data and what the factor expects?
This question is better asked on the Google Group since it is a custom code question and not a library issue. Please send your query to the Google Group.

closed 2024-07-12 by varunagrawal

dolgae03, 2024-07-12:
Thank you for your advice I will join google group.