borglab/gtsam #1779: Modernize GTSAM containers

acxz, 2024-07-09:
## Feature

Need some GTSAM containers, noticed the containers we provide are pretty outdated ([latest push to Docker Hub 3 years ago](https://hub.docker.com/r/borglab/ubuntu-gtsam/tags) and based off bionic).

I also noticed some todo's to use the [Compose Spec](https://compose-spec.io/) and noticed that our list of images could be streamlined with the use of tags.

tldr: modernize containers with newer versions, better tooling (compose), use of tags for various configurations, slimmer containers with best practices of writing container files.

## Motivation

Need newer GTSAM images, smaller image sizes, and easier building/pushing to online registries

## Pitch

Replace build.sh scripts with compose.yaml

Use `.env` files to specify various gtsam configurations, such as base ubuntu image, tbb on/off, python on/off, gtsam git tag.
Have a single gtsam image repository (instead of multiple potentially unused intermediary images) with various tags corresponding to build configurations
Have another image for the gtsam-vnc image (of course tagged with a base gtsam image)

Create a simple script to push various tagged images to docker hub in one go.

Create smaller image sizes, by condensing `RUN` commands and using appropriate flags such as [`--no-install-recommends`](https://ubuntu.com/blog/we-reduced-our-docker-images-by-60-with-no-install-recommends).

## Alternatives

N/A

## Additional context

For full transparency, I'm writing this issue after I have implemented a resolution for it.

acxz, 2024-07-11:
@varunagrawal I know this issue got autoclosed, but could we keep it open until the images are pushed to Docker Hub and/or we decide on where the containers related files should reside (in the main repo or in the `docker-images` repo) and clean up the duplicate images in both this repo and the `docker-images` repo.

varunagrawal, 2024-07-12:
Oh yeah absolutely. Thanks for being on top of this. 

I'm waiting on Docker to renew our OSS license so we can run autobuilds. That may take a while.

acxz, 2024-07-12:
Running Docker autobuilds may actually require some changes to our image building. I don't how it would handle our multiconfiguration stuff (i.e. with the `.env` file and `hub_push.sh` script).

My PR was based on local building and pushing.
Would it be possible for you to simply run the `hub_push.sh` script locally and get the images out on Docker Hub?

I apologize if I'm being antsy, just want to see the Docker Hub updated. I'm totally okay with waiting if autobuilds is the way to go.

varunagrawal, 2024-07-12:
Built and pushed the images: https://hub.docker.com/repository/docker/borglab/gtsam/general

attn: @dellaert

PR #1780 (2024-07-11): modernize containers
use compose spec, use single image repo with tags for various configs, create a docker hub push script, clean up/slim down container files

resolves #1779
closed 2024-07-12 by varunagrawal