borglab/gtsam #1783: Failed to import after compiling with build type `Timing`

Luke20000429, 2024-07-11:
## Description

I built gtsam with 
```
cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.11  -DGTSAM_ENABLE_BOOST_SERIALIZATION=OFF -DGTSAM_USE_BOOST_FEATURES=ON -DCMAKE_BUILD_TYPE=Timing -DGTSAM_WITH_TBB=OFF
```
I want to see the duration breakdown of each step. I can run cpp examples and see the timing results, but when I try to run my python code, I got an import error
```
 import gtsam
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/username/.local/lib/python3.11/site-packages/gtsam/__init__.py", line 9, in <module>
    from gtsam import gtsam, utils
ImportError: cannot import name 'gtsam' from partially initialized module 'gtsam' (most likely due to a circular import) (/home/username/.local/lib/python3.11/site-packages/gtsam/__init__.py)
```
I didn't see this error when building with `Release`

## Expected behavior

## Environment
gcc-9.4, ubuntu 20.04, boost 1.82.0, python 3.11

varunagrawal, 2024-09-26:
Can you try setting `-DGTSAM_ENABLE_BOOST_SERIALIZATION=ON`?
The wrapper depends on Boost serialization more than it does on Boost features.

varunagrawal, 2024-09-26:
Okay, can confirm this is a bug. Making a fix.

PR #1849 (2024-09-27): Remove postfix for compiled Python library
When compiling with Timing and Profiling, the extra postfix gets added which causes import errors.

Fixes #1783.

Also added a missing guard and cleaned up the CMakeLists.txt file a bit.
closed 2024-09-27 by varunagrawal