borglab/gtsam #1784: Add "point" method to Python realization of SmartProjectionFactor

fatrybl, 2024-07-12:
## Feature

Hello, there is a C++ example https://github.com/borglab/gtsam/blob/develop/examples/SFMExample_SmartFactor.cpp#L51 where the SmartProjectionPoseFactor is used. There is a SmartProjectionPoseFactor->point method which returns landmark 3D position. However, this method does not exist in Python realization of SmartProjectionPose3Factor. Could you please add it ?

## Motivation

This would make life much easier if one could get 3D position of a landmark to visualize without extra code.

## Pitch

## Alternatives

## Additional context

ProfFan, 2024-08-14:
You can easily add it in the `.i` files and submit a PR :)

varunagrawal, 2024-08-19:
The `SmartProjectionPose3Factor.point` method has been wrapped and available since October 2023. @fatrybl are you using the latest version of GTSAM?

Simple example which works for me:
```python
import gtsam
measurementNoise = gtsam.noiseModel.Isotropic.Sigma(2, 1.0)
K = gtsam.Cal3_S2(50.0, 50.0, 0.0, 50.0, 50.0)
f = gtsam.SmartProjectionPose3Factor(measurementNoise, K)
f.point()
```

closed 2024-08-19 by varunagrawal

fatrybl, 2024-09-04:
@varunagrawal 
Hello, i`m using gtsam 4.2
The provided example does not work for me.
[image]

fatrybl, 2024-12-16:
The issue is no longer relevant if building the  GTSAM  with python-wrapper from source (latest current develop). Hope the fix will be added to the upcoming gtsam 4.3 release (with pip wheel)