borglab/gtsam #1787: ROS2 Lookup error

js-fwr, 2024-07-24:
## Description

I created a ROS2 Node for factorgraph sensor fusion. I build GTsam 4.2 and am able to run the examples provided without any problem. I also can build the ros2 Node using colcon, but as soon as i run it i receive an symbol lookup error.  
`/ros2_ws/install/factor_graph_fusion/lib/factor_graph_fusion/factor_graph_fusion_node: symbol lookup error: /home/johannes/ros2_ws/install/factor_graph_fusion/lib/factor_graph_fusion/factor_graph_fusion_node: undefined symbol: _ZN5gtsam5ISAM26updateERKNS_20NonlinearFactorGraphERKNS_6ValuesERKSt6vectorImN3tbb6detail2d113tbb_allocatorImEEERKN5boost8optionalINS_7FastMapImiEEEERKNSH_INS_8FastListImEEEESR_b
[ros2run]: Process exited with failure 127
`

## Steps to reproduce

1. include gtsam headers in ros2 node
2.add GTSAM to Cmakefile and package.xml ( files attached)
3. build and run the node

## Expected behavior

Since colcon build is running correctly and examples are also running as expected, i assume the node to run the includes as supposed.

## Environment

OS: Ubuntu 22.04.4 LTS
ROS2 Version: Humble
GTSAM-Version 4.2 

## Additional information

[image]

[CMakeLists.txt](https://github.com/user-attachments/files/16363001/CMakeLists.txt)

jlblancoc, 2024-07-24:
It looks a problem with LD_LIBRARY_PATH, right?
Have you tried installing gtsam from apt so colcon includes its path in LD_LIBRARY_PATH?
You can manually append to LD_LIBRARY_PATH just to verify that then your node runs, but that's not a proper fix. 

Sometimes, I found similar errors after manually running "make" or "make install" under `build/foo` after running `colcon build...`. That seems to cause confusion to all the colcon+ament machinery. The solution was to clear env vars (open a new terminal), `rm -fr install build`, and run `colcon` again. Of course, place gtsam under `ros2_ws/src` so it's handled as another ROS2 package by colcon.

Pay attention to differences in gtsam/package.xml between the `4.2` branch (the only one officially released to ROS 2 build farms, and which I have verified that builds **and run** as expected), and `develop`, and report if you find suspicious differences.

js-fwr, 2024-07-25:
thanks for ur suggestions. i was able to resolve the issue following ure advice: 

> Sometimes, I found similar errors after manually running "make" or "make install" under build/foo after running colcon build.... That seems to cause confusion to all the colcon+ament machinery. The [...]

closed 2024-07-27 by jlblancoc