borglab/gtsam #1799: [MatlabWrapper] Ignoring classes: ['gtsam::Point2', 'gtsam::Point3', 'gtsam::CustomFactor']

DanielSofia, 2024-08-18:
Hi, I'm want to install GTSAM Matlab toolbox. Using Matalb 2023a, Ubuntu 22.04 LTS. ALways I try to "make" inside appears the following errors:
 [84%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/geometry/Event.cpp.o
[MatlabWrapper] Ignoring classes: ['gtsam::Point2', 'gtsam::Point3', 'gtsam::CustomFactor']

after some compilation progress:
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/AHRS.cpp.o
/home/danielsofia/Downloads/gtsam/build/wrap/gtsam/gtsam_wrapper.cpp: In function ‘void gtsamCal3_S2Stereo_vector_1468(int, mxArray**, int, const mxArray**)’:
/home/danielsofia/Downloads/gtsam/build/wrap/gtsam/gtsam_wrapper.cpp:18824:45: error: ‘Vector6’ was not declared in this scope; did you mean ‘gtsam::Vector6’?
18824 |   out[0] = wrap_shared_ptr(std::make_shared<Vector6>(obj->vector()),"Vector6", false);
      |                                             ^~~~~~~
      |                                             gtsam::Vector6
In file included from /home/danielsofia/Downloads/gtsam/wrap/matlab.h:25,
                 from /home/danielsofia/Downloads/gtsam/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/home/danielsofia/Downloads/gtsam/gtsam/base/Vector.h:51:9: note: ‘gtsam::Vector6’ declared here
   51 |   using Vector##N = Eigen::Matrix<double, N, 1>; \
      |         ^~~~~~
/home/danielsofia/Downloads/gtsam/gtsam/base/Vector.h:56:1: note: in expansion of macro ‘GTSAM_MAKE_VECTOR_DEFS’
   56 | GTSAM_MAKE_VECTOR_DEFS(6)
      | ^~~~~~~~~~~~~~~~~~~~~~
/home/danielsofia/Downloads/gtsam/build/wrap/gtsam/gtsam_wrapper.cpp:18824:53: error: no matching function for call to ‘make_shared<<expression error> >(gtsam::Vector6)’
18824 |   out[0] = wrap_shared_ptr(std::make_shared<Vector6>(obj->vector()),"Vector6", false);
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from /usr/include/c++/11/memory:77,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/Testable.h:40,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/Manifold.h:23,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/Lie.h:26,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/VectorSpace.h:11,
                 from /home/danielsofia/Downloads/gtsam/gtsam/geometry/Point2.h:20,
                 from /home/danielsofia/Downloads/gtsam/wrap/matlab.h:27,
                 from /home/danielsofia/Downloads/gtsam/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/c++/11/bits/shared_ptr.h:875:5: note: candidate: ‘template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)’
  875 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
/usr/include/c++/11/bits/shared_ptr.h:875:5: note:   template argument deduction/substitution failed:
/home/danielsofia/Downloads/gtsam/build/wrap/gtsam/gtsam_wrapper.cpp:18824:53: error: template argument 1 is invalid
18824 |   out[0] = wrap_shared_ptr(std::make_shared<Vector6>(obj->vector()),"Vector6", false);
      |                            ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from /home/danielsofia/Downloads/gtsam/gtsam/3rdparty/Eigen/Eigen/Core:337,
                 from /home/danielsofia/Downloads/gtsam/gtsam/3rdparty/Eigen/Eigen/Dense:1,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/OptionalJacobian.h:24,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/Matrix.h:27,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/Manifold.h:22,
                 from /home/danielsofia/Downloads/gtsam/gtsam/base/GenericValue.h:22,
                 from /home/danielsofia/Downloads/gtsam/gtsam/nonlinear/Values.h:29,
                 from /home/danielsofia/Downloads/gtsam/gtsam/nonlinear/NonlinearFactor.h:24,
                 from /home/danielsofia/Downloads/gtsam/gtsam_unstable/nonlinear/LinearizedFactor.h:22,
                 from /home/danielsofia/Downloads/gtsam/gtsam_unstable/nonlinear/LinearizedFactor.cpp:18:
/home/danielsofia/Downloads/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/products/SelfadjointMatrixVector.h: In function ‘static void Eigen::internal::selfadjoint_product_impl<Lhs, LhsMode, false, Rhs, 0, true>::run(Dest&, const Lhs&, const Rhs&, const Scalar&) [with Dest = Eigen::Matrix<double, -1, 1>; Lhs = Eigen::Block<const Eigen::Matrix<double, -1, -1>, -1, -1, false>; int LhsMode = 18; Rhs = Eigen::Matrix<double, -1, 1>]’:
/home/danielsofia/Downloads/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/products/SelfadjointMatrixVector.h:229:7: warning: ‘result’ may be used uninitialized [-Wmaybe-uninitialized]
  227 |     internal::selfadjoint_matrix_vector_product<Scalar, Index, (internal::traits<ActualLhsTypeCleaned>::Flags&RowMajorBit) ? RowMajor : ColMajor,
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  228 |                                                 int(LhsUpLo), bool(LhsBlasTraits::NeedToConjugate), bool(RhsBlasTraits::NeedToConjugate)>::run
      |                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  229 |       (
      |       ^
  230 |         lhs.rows(),                             // size
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  231 |         &lhs.coeffRef(0,0),  lhs.outerStride(), // lhs info
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  232 |         actualRhsPtr,                           // rhs info
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  233 |         actualDestPtr,                          // result info
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  234 |         actualAlpha                             // scale factor
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  235 |       );
      |       ~
/home/danielsofia/Downloads/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/products/SelfadjointMatrixVector.h:41:6: note: by argument 4 of type ‘const double*’ to ‘static void Eigen::internal::selfadjoint_matrix_vector_product<Scalar, Index, StorageOrder, UpLo, ConjugateLhs, ConjugateRhs, Version>::run(Index, const Scalar*, Index, const Scalar*, Scalar*, Scalar) [with Scalar = double; Index = long int; int StorageOrder = 0; int UpLo = 2; bool ConjugateLhs = false; bool ConjugateRhs = false; int Version = 0]’ declared here
   41 | void selfadjoint_matrix_vector_product<Scalar,Index,StorageOrder,UpLo,ConjugateLhs,ConjugateRhs,Version>::run(
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 235 lines cut ...]
  335 |             dest.data(),dest.innerStride(),
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  336 |             actualAlpha);
      |             ~~~~~~~~~~~~
/home/danielsofia/Downloads/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/products/TriangularMatrixVector.h:105:24: note: by argument 5 of type ‘const double*’ to ‘static void Eigen::internal::triangular_matrix_vector_product<Index, Mode, LhsScalar, ConjLhs, RhsScalar, ConjRhs, 1, Version>::run(Index, Index, const LhsScalar*, Index, const RhsScalar*, Index, Eigen::internal::triangular_matrix_vector_product<Index, Mode, LhsScalar, ConjLhs, RhsScalar, ConjRhs, 1, Version>::ResScalar*, Index, const ResScalar&) [with Index = long int; int Mode = 6; LhsScalar = double; bool ConjLhs = false; RhsScalar = double; bool ConjRhs = false; int Version = 0]’ declared here
  105 | EIGEN_DONT_INLINE void triangular_matrix_vector_product<Index,Mode,LhsScalar,ConjLhs,RhsScalar,ConjRhs,RowMajor,Version>
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 93%] Building CXX object python/CMakeFiles/gtsam_py.dir/slam.cpp.o
[ 93%] Linking CXX executable ISAM2_SmartFactorStereo_IMU
[ 93%] Built target ISAM2_SmartFactorStereo_IMU
[ 93%] Building CXX object python/CMakeFiles/gtsam_py.dir/sfm.cpp.o
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:96: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:33372: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 93%] Building CXX object python/CMakeFiles/gtsam_py.dir/navigation.cpp.o
[ 95%] Building CXX object python/CMakeFiles/gtsam_py.dir/basis.cpp.o
[ 95%] Building CXX object python/CMakeFiles/gtsam_py.dir/hybrid.cpp.o
[ 95%] Linking CXX shared module gtsam/gtsam.cpython-310-x86_64-linux-gnu.so
lto-wrapper: warning: using serial compilation of 128 LTRANS jobs
[ 97%] Built target gtsam_py
make: *** [Makefile:166: all] Error 2

And all finished.

CAn you help me please? Thanks a lot

varunagrawal, 2024-08-19:
@DanielSofia thank you for reporting this. This is my mistake and I'll make a PR to rectify this in a minute.

DanielSofia, 2024-08-19:
@varunagrawal thanks a lot. Please, when you can change the code, please tell me. Thanks again 👍

varunagrawal, 2024-08-20:
@DanielSofia the PR has been merged, so if you pull the latest `develop` branch, it should compile for you. Please let me know if there is anything else that comes up.

DanielSofia, 2024-08-23:
@varunagrawal Sorry but I tried to compile in different ways and methods but the error  "Ignoring classes: ['gtsam::Point2', 'gtsam::Point3', 'gtsam::CustomFactor'] appears in the same way, and in the end I can't install the GTSAM Matlab Toolbox.

[ 88%] Generating ../wrap/gtsam/gtsam_wrapper.cpp
[MatlabWrapper] Ignoring classes: ['gtsam::Point2', 'gtsam::Point3', 'gtsam::CustomFactor']
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/geometry/BearingS2.cpp.o
[ 88%] Linking CXX executable easyPoint2KalmanFilter
[ 88%] Built target easyPoint2KalmanFilter
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/geometry/Event.cpp.o
[ 88%] Linking CXX executable TriangulationLOSTExample
[ 88%] Built target TriangulationLOSTExample
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/geometry/Pose3Upright.cpp.o
[ 88%] Linking CXX executable VisualISAM2Example
[ 88%] Built target VisualISAM2Example
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/geometry/SimPolygon2D.cpp.o
[ 88%] Linking CXX executable SFMExample_SmartFactorPCG
[ 88%] Built target SFMExample_SmartFactorPCG
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/geometry/SimWall2D.cpp.o
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/linear/LPInitSolver.cpp.o
[ 88%] Linking CXX executable VisualISAMExample
[ 88%] Built target VisualISAMExample
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/linear/LPSolver.cpp.o
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/linear/QPSParser.cpp.o
[ 88%] Linking CXX executable SFMExample_SmartFactor
[ 88%] Built target SFMExample_SmartFactor
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/linear/QPSolver.cpp.o
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/discrete/AllDiff.cpp.o
[ 88%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/discrete/CSP.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/discrete/Domain.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/discrete/Scheduler.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/discrete/SingleValue.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/dynamics/PoseRTV.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/dynamics/SimpleHelicopter.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/ConcurrentBatchFilter.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/ConcurrentBatchSmoother.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/ConcurrentFilteringAndSmoothing.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/ConcurrentIncrementalFilter.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/ConcurrentIncrementalSmoother.cpp.o
[ 90%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/IncrementalFixedLagSmoother.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/nonlinear/LinearizedFactor.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/AHRS.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/DummyFactor.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/LocalOrientedPlane3Factor.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/Mechanization_bRn2.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/ProjectionFactorRollingShutter.cpp.o
[ 92%] Building CXX object gtsam_unstable/CMakeFiles/gtsam_unstable.dir/slam/RelativeElevationFactor.cpp.o
In file included from /home/danielsofia/Downloads/gtsam-develop/gtsam/3rdparty/Eigen/Eigen/Core:337,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/3rdparty/Eigen/Eigen/Dense:1,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/base/OptionalJacobian.h:24,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/base/Matrix.h:27,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/base/Manifold.h:22,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/base/GenericValue.h:22,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/nonlinear/Values.h:29,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam/nonlinear/NonlinearFactor.h:24,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam_unstable/nonlinear/LinearizedFactor.h:22,
                 from /home/danielsofia/Downloads/gtsam-develop/gtsam_unstable/nonlinear/LinearizedFactor.cpp:18:
/home/danielsofia/Downloads/gtsam-develop/gtsam/3rdparty/Eigen/Eigen/src/Core/products/SelfadjointMatrixVector.h: In function ‘static void Eigen::internal::selfadjoint_product_impl<Lhs, LhsMode, false, Rhs, 0, true>::run(Dest&, const Lhs&, const Rhs&, const Scalar&) [with Dest = Eigen::Matrix<double, -1, 1>; Lhs = Eigen::Block<const Eigen::Matrix<double, -1, -1>, -1, -1, false>; int LhsMode = 18; Rhs = Eigen::Matrix<double, -1, 1>]’:
[... cut ...]
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:96: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:33082: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

There is something I missed install? I strictly used the following guide: https://github.com/borglab/gtsam/blob/develop/matlab/README.md
If you can help I always be grateful to your work.
Thanks again @varunagrawal

closed 2024-08-25 by varunagrawal

varunagrawal, 2024-08-25:
@DanielSofia can you try now? It should work.

DanielSofia, 2024-08-27:
@varunagrawal I already try and all works 100%. Now I can compile gtsam wrapper for MATLAB Toolbox in Ubuntu 22.04 LTS. Thanks a lot for your time and attention.

PR #1800 (2024-08-20): Fix `geometry.i` namespace issue
I forgot to update the namespace for a method in `geometry.i` which is leading to the Matlab wrapper not compiling.
Fixes #1799.