borglab/gtsam #1806: Integrate compass heading or magnetometer constrain from 9 axis IMU using the GTSAM Factor graph framework

gjf281, 2024-08-23:
## Feature

How to integrate compass heading or magnetometer constrain from 9 axis IMU using the GTSAM Factor graph framework?
 

## Motivation

## Pitch

## Alternatives

## Additional context

ProfFan, 2024-08-23:
Please check `MagFactor` etc., if you are talking about the algorithm, you will need to read the cited papers there.

closed 2024-08-23 by ProfFan

gjf281, 2024-08-27:
Thanks for your reply very much.
I have a further problem. I found a reference code that uses MagFactor class：

`#include <gtsam/navigation/MagFactor.h>
#include <gtsam/navigation/ImuFactor.h>
#include <gtsam/navigation/CombinedImuFactor.h>
#include <gtsam/navigation/GPSFactor.h>
#include <gtsam/navigation/PreintegratedCombinedMeasurements.h>
#include <gtsam/slam/PriorFactor.h>
#include <gtsam/nonlinear/ISAM2.h>
#include <gtsam/nonlinear/LevenbergMarquardtOptimizer.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>

using namespace gtsam;

int main() {
    // 创建非线性因子图
    NonlinearFactorGraph graph;

    // 添加先验因子
    Pose3 priorPose(Rot3::Quaternion(1, 0, 0, 0), Point3(0, 0, 0));
    auto priorNoise = noiseModel::Diagonal::Sigmas((Vector(6) << 0.3, 0.3, 0.3, 0.1, 0.1, 0.1).finished());
    graph.emplace_shared<PriorFactor<Pose3>>(1, priorPose, priorNoise);

    // 添加磁力计因子
    Vector3 measuredMagneticField(0.3, 0.0, 0.5);  // 假设这是测量到的磁场
    auto magNoise = noiseModel::Isotropic::Sigma(3, 0.1);
    graph.emplace_shared<MagFactor>(1, measuredMagneticField, magNoise);

    // 创建初始值
    Values initialEstimate;
    initialEstimate.insert(1, Pose3(Rot3::Quaternion(1, 0, 0, 0), Point3(0, 0, 0)));

    // 优化
    LevenbergMarquardtOptimizer optimizer(graph, initialEstimate);
    Values result = optimizer.optimize();

    // 输出结果
    result.print("Final Result:\n");

    return 0;
}`

I use the xsens MTi-630 IMU. Its output of magnetometer measurement is quaternion：
`header: 
  seq: 44002
  stamp: 
    secs: 1626894399
    nsecs: 853138584
  frame_id: "imu_link"
orientation: 
  x: 0.01064246241
  y: -0.0552232116461
  z: -0.991362392902
  w: -0.118481434882
`
The question is how to use MagFactor if  the magnetometer measurement is quaternion?