borglab/gtsam #1819: Ubuntu nightly packaging fails with cmake error, likely due to PR #1818

berndpfrommer, 2024-08-30:
The Ubuntu nightly builds (on Ubuntu 22.04) fail with the following error:
```
CMake Error at CMakeLists.txt:4 (cmake_policy):
  Policy "CMP0156" is not known to this version of CMake.

CMake Error at CMakeLists.txt:5 (cmake_policy):
  Policy "CMP0167" is not known to this version of CMake.
```
[Here](https://launchpadlibrarian.net/746574457/buildlog_ubuntu-jammy-amd64.gtsam_4.1.0-1ubuntu2~2182.gbp158895_BUILDING.txt.gz) is the log of the failed build.

This is very likely related to PR #1818.

Policy CMP0156 was introduced with CMake 3.29
Policy CMP0167 was introduced with CMake 3.30

I cannot figure out how to check if a policy exists before setting it. The pedestrian fix would probably look something like:
```
if(CMAKE_VERSION VERSION_GREATER_EQUAL "3.29.0") 
  cmake_policy(SET CMP0156 NEW)
endif()

if(CMAKE_VERSION VERSION_GREATER_EQUAL "3.30.0") 
  cmake_policy(SET CMP0167 OLD)
endif()
```

dellaert, 2024-08-30:
Apologies for the break. I can make a PR with whichever solution you prefer.

berndpfrommer, 2024-08-30:
Thanks!
I think the second fix by @jlblancoc is better.

ProfFan, 2024-09-03:
This probably can be closed?

jlblancoc, 2024-09-03:
Yes, it builds now on the PPA and the ROS build farms :+1:

closed 2024-09-03 by jlblancoc