borglab/gtsam #1827: more then 5348 nodes,optimize crash

luyuanxunnuli, 2024-09-07:
code in gtsam example Pose3Localization:

int main(const int argc, const char* argv[]) {
  // Read graph from file
  string g2oFile;
  if (argc < 2)
    //g2oFile = findExampleDataFile("pose3Localizationexample.txt");
      g2oFile = findExampleDataFile("data.txt");
  else
    g2oFile = argv[1];

  NonlinearFactorGraph::shared_ptr graph;
  Values::shared_ptr initial;
  bool is3D = true;
  boost::tie(graph, initial) = readG2o(g2oFile, is3D);

  // Add prior on the first key
  auto priorModel = noiseModel::Diagonal::Variances(
      (Vector(6) << 1e-6, 1e-6, 1e-6, 1e-4, 1e-4, 1e-4).finished());
  Key firstKey = 0;
  for (const auto key : initial->keys()) {
    std::cout << "Adding prior to g2o file " << std::endl;
    firstKey = key;
    graph->addPrior(firstKey, Pose3(), priorModel);
    break;
  }

  std::cout << "Optimizing the factor graph" << std::endl;
  GaussNewtonParams params;
  params.setVerbosity("TERMINATION");  // show info about stopping conditions
  GaussNewtonOptimizer optimizer(*graph, *initial, params);
  Values result = optimizer.optimize();
  std::cout << "Optimization complete" << std::endl;

  return 0;

  std::cout << "initial error=" << graph->error(*initial) << std::endl;
  std::cout << "final error=" << graph->error(result) << std::endl;

  if (argc < 3) {
    result.print("result");
  } else {
    const string outputFile = argv[2];
    std::cout << "Writing results to file: " << outputFile << std::endl;
    NonlinearFactorGraph::shared_ptr graphNoKernel;
    Values::shared_ptr initial2;
    boost::tie(graphNoKernel, initial2) = readG2o(g2oFile);
    writeG2o(*graphNoKernel, result, outputFile);
    std::cout << "done! " << std::endl;
  }

  // Calculate and print marginal covariances for all variables
  Marginals marginals(*graph, result);
  for (const auto& key_pose : result.extract<Pose3>()) {
    std::cout << marginals.marginalCovariance(key_pose.first) << endl;
  }
  return 0;
}

data.txt is my data,like this:
VERTEX_SE3:QUAT 0 0 0 0 0 0 0 0
VERTEX_SE3:QUAT 1 1 0 0 0 0 0 0
VERTEX_SE3:QUAT 2 2 0 0 0 0 0 0
VERTEX_SE3:QUAT 3 3 0 0 0 0 0 0
VERTEX_SE3:QUAT 4 4 0 0 0 0 0 0
VERTEX_SE3:QUAT 5 5 0 0 0 0 0 0
VERTEX_SE3:QUAT 6 6 0 0 0 0 0 0
VERTEX_SE3:QUAT 7 7 0 0 0 0 0 0
VERTEX_SE3:QUAT 8 8 0 0 0 0 0 0
VERTEX_SE3:QUAT 9 9 0 0 0 0 0 0
VERTEX_SE3:QUAT 10 10 0 0 0 0 0 0
VERTEX_SE3:QUAT 11 11 0 0 0 0 0 0
VERTEX_SE3:QUAT 12 12 0 0 0 0 0 0
VERTEX_SE3:QUAT 13 13 0 0 0 0 0 0
VERTEX_SE3:QUAT 14 14 0 0 0 0 0 0
VERTEX_SE3:QUAT 15 15 0 0 0 0 0 0
........
EDGE_SE3:QUAT 0 1 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 1 2 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 2 3 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 3 4 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 4 5 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 5 6 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 6 7 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 7 8 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 8 9 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 9 10 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 10 11 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 11 12 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 12 13 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 13 14 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 14 15 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 15 16 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 16 17 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
EDGE_SE3:QUAT 17 18 1 0 0 0.01 0.01 0.01 0.01 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001 0.001
.....

10000 nodes  9999 edges

luyuanxunnuli, 2024-09-11:
First of all, I do a apologize. Today, I debugged and ran the official Pose3Localization directly on Visual Studio, only replacing loacal.txe with my own data.txt and setting the stack size to be larger. It did indeed crash, but when I replaced the GaussNewtonOptimizer with the LM optimizer, it  run normally. I replied to you yesterday that The LM optimizer with larger stack size crash was incorrect.GaussNewtonOptimizer  crashes even if the data is small. @ProfFan

ProfFan, 2024-09-12:
So it's the G-N optimizer that is failing? @hunkyu is this the same case for you?

I am thinking about an undefined behavior somewhere in the G-N code as it is indeed less battle-tested. But I have no bandwidth now with paper deadlines, so any help is appreciated. Ideally if you could run with ASAN and put the log here...

graphic-zhang, 2025-01-07:
more than 13500 node , i crash on ubuntu too 
 gtsam::internal::optimizeInPlace(boost::shared_ptr<gtsam::ISAM2Clique> const&, gtsam::VectorValues*) () from /usr/local/lib/libgtsam.so.4
gtsam::DeltaImpl::UpdateGaussNewtonDelta(std::vector<boost::shared_ptr<gtsam::ISAM2Clique>, tbb::tbb_allocator<boost::shared_ptr<gtsam::ISAM2Clique> > > const&, gtsam::FastSet<unsigned long> const&, double, gtsam::VectorValues*) () from /usr/local/lib/libgtsam.so.4