borglab/gtsam #1855: Cannot compile matlab toolbox (wrapper)

Huang-Chuan, 2024-09-30:
## Description

Cannot compile matlab toolbox but it compiles with no error when toolbox option is off. Seems to have some incompatibility with boost?

## Environment

Ubuntu 22.04, Boost version 1.74.0, MATLAB 2024b

## Additional information
[ 97%] Building CXX object matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values.h:571,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/NonlinearFactor.h:24,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/FunctorizedFactor.h:21,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/basis/BasisFactors.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:14:
/home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values-inl.h: In member function ‘gtsam::Values::Filtered<gtsam::Value> gtsam::Values::filter(const std::function<bool(long unsigned int)>&)’:
/home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values-inl.h:253:25: warning: ‘gtsam::Values::Filtered<ValueType> gtsam::Values::filter(const std::function<bool(long unsigned int)>&) [with ValueType = gtsam::Value]’ is deprecated [-Wdeprecated-declarations]
  253 |     return filter<Value>(filterFcn);
      |            ~~~~~~~~~~~~~^~~~~~~~~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/NonlinearFactor.h:24,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/FunctorizedFactor.h:21,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/basis/BasisFactors.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:14:
/home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values.h:421:5: note: declared here
  421 |     filter(const std::function<bool(Key)>& filterFcn = &_truePredicate<Key>);
      |     ^~~~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values.h:571,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/NonlinearFactor.h:24,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/FunctorizedFactor.h:21,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/basis/BasisFactors.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:14:
/home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values-inl.h: In member function ‘gtsam::Values::ConstFiltered<gtsam::Value> gtsam::Values::filter(const std::function<bool(long unsigned int)>&) const’:
/home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values-inl.h:265:25: warning: ‘gtsam::Values::ConstFiltered<ValueType> gtsam::Values::filter(const std::function<bool(long unsigned int)>&) const [with ValueType = gtsam::Value]’ is deprecated [-Wdeprecated-declarations]
  265 |     return filter<Value>(filterFcn);
      |            ~~~~~~~~~~~~~^~~~~~~~~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/NonlinearFactor.h:24,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/FunctorizedFactor.h:21,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/basis/BasisFactors.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:14:
/home/chuan/Downloads/gtsam-4.2/gtsam/nonlinear/Values.h:429:47: note: declared here
  429 |     ConstFiltered<ValueType> GTSAM_DEPRECATED filter(
      |                                               ^~~~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteBayesNet.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:22:
/home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteDistribution.h: In member function ‘size_t gtsam::DiscreteDistribution::solve() const’:
/home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteDistribution.h:99:56: warning: ‘size_t gtsam::DiscreteConditional::solve(const gtsam::DiscreteValues&) const’ is deprecated [-Wdeprecated-declarations]
   99 |   size_t GTSAM_DEPRECATED solve() const { return Base::solve({}); }
      |                                                        ^~~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteBayesNet.h:21,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:22:
/home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteConditional.h:269:27: note: declared here
  269 |   size_t GTSAM_DEPRECATED solve(const DiscreteValues& parentsValues) const;
      |                           ^~~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteBayesNet.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:22:
/home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteDistribution.h:99:61: warning: ‘size_t gtsam::DiscreteConditional::solve(const gtsam::DiscreteValues&) const’ is deprecated [-Wdeprecated-declarations]
   99 |   size_t GTSAM_DEPRECATED solve() const { return Base::solve({}); }
      |                                                  ~~~~~~~~~~~^~~~
In file included from /home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteBayesNet.h:21,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:22:
/home/chuan/Downloads/gtsam-4.2/gtsam/discrete/DiscreteConditional.h:269:27: note: declared here
  269 |   size_t GTSAM_DEPRECATED solve(const DiscreteValues& parentsValues) const;
      |                           ^~~~~
In file included from /usr/include/boost/smart_ptr/make_shared.hpp:14,
                 from /usr/include/boost/make_shared.hpp:14,
                 from /home/chuan/Downloads/gtsam-4.2/wrap/matlab.h:41,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/boost/smart_ptr/make_shared_object.hpp: In instantiation of ‘typename boost::detail::sp_if_not_array<T>::type boost::make_shared(Args&& ...) [with T = std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor>; Args = {std::pair<boost::shared_ptr<gtsam::DiscreteConditional>, boost::shared_ptr<gtsam::DecisionTreeFactor> >}; typename boost::detail::sp_if_not_array<T>::type = boost::shared_ptr<std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor> >]’:
/home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:12189:111:   required from here
**/usr/include/boost/smart_ptr/make_shared_object.hpp:256:5: error: no matching function for call to ‘std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor>::pair(std::pair<boost::shared_ptr<gtsam::DiscreteConditional>, boost::shared_ptr<gtsam::DecisionTreeFactor> >)’
[... 346 lines cut ...]
                 from /usr/include/c++/11/bits/ios_base.h:41,
                 from /usr/include/c++/11/streambuf:41,
                 from /usr/include/c++/11/bits/streambuf_iterator.h:35,
                 from /usr/include/c++/11/iterator:66,
                 from /usr/include/boost/concept_check.hpp:22,
                 from /usr/include/boost/range/concepts.hpp:19,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/base/types.h:24,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/global_includes.h:22,
                 from /home/chuan/Downloads/gtsam-4.2/gtsam/base/Vector.h:28,
                 from /home/chuan/Downloads/gtsam-4.2/wrap/matlab.h:25,
                 from /home/chuan/Downloads/gtsam-4.2/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/c++/11/bits/stl_pair.h:315:17: note: candidate: ‘std::pair<_T1, _T2>::pair(std::pair<_T1, _T2>&&) [with _T1 = gtsam::DiscreteConditional; _T2 = gtsam::DecisionTreeFactor]’
  315 |       constexpr pair(pair&&) = default;         ///< Move constructor
      |                 ^~~~
/usr/include/c++/11/bits/stl_pair.h:315:22: note:   no known conversion for argument 1 from ‘std::pair<boost::shared_ptr<gtsam::DiscreteConditional>, boost::shared_ptr<gtsam::DecisionTreeFactor> >’ to ‘std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor>&&’
  315 |       constexpr pair(pair&&) = default;         ///< Move constructor
      |                      ^~~~~~
/usr/include/c++/11/bits/stl_pair.h:314:17: note: candidate: ‘std::pair<_T1, _T2>::pair(const std::pair<_T1, _T2>&) [with _T1 = gtsam::DiscreteConditional; _T2 = gtsam::DecisionTreeFactor]’
  314 |       constexpr pair(const pair&) = default;    ///< Copy constructor
      |                 ^~~~
[... cut ...]  boost::shared_ptr<gtsam::DecisionTreeFactor> >’ to ‘const std::pair<gtsam::DiscreteConditional, gtsam::DecisionTreeFactor>&’
  314 |       constexpr pair(const pair&) = default;    ///< Copy constructor
      |                      ^~~~~~~~~~~
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:95: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:32558: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2

varunagrawal, 2024-10-08:
Are you using the 4.2 release?

Huang-Chuan, 2024-10-09:
Yes.

varunagrawal, 2024-10-10:
I'm not sure why the release version shouldn't compile. Can you try the latest develop version? That will definitely compile (tried it out myself last night).

varunagrawal, 2024-10-14:
@Huang-Chuan any updates?

Huang-Chuan, 2024-10-15:
@varunagrawal I will try to compile it next week, because right now I am occupied on a conference.

Huang-Chuan, 2024-10-24:
@varunagrawal I downloaded the latest develop version and compiled it. It failed again, the error is different, though.

[100%] Building CXX object matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o
/home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp: In function ‘void gtsamPCGSolverParameters_set_preconditioner_2497(int, mxArray**, int, const mxArray**)’:
/home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:28709:26: error: no match for ‘operator=’ (operand types are ‘std::shared_ptr<gtsam::PreconditionerParameters>’ and ‘std::__shared_ptr_access<gtsam::PreconditionerParameters, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘gtsam::PreconditionerParameters’})
28709 |   obj->preconditioner = *preconditioner;
      |                          ^~~~~~~~~~~~~~
In file included from /usr/include/c++/9/memory:81,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Testable.h:40,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Manifold.h:23,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Lie.h:26,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/VectorSpace.h:11,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/geometry/Point2.h:20,
                 from /home/chuan/Desktop/gtsam-develop/wrap/matlab.h:27,
                 from /home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/c++/9/bits/shared_ptr.h:309:19: note: candidate: ‘std::shared_ptr<_Tp>& std::shared_ptr<_Tp>::operator=(const std::shared_ptr<_Tp>&) [with _Tp = gtsam::PreconditionerParameters]’
  309 |       shared_ptr& operator=(const shared_ptr&) noexcept = default;
      |                   ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:309:29: note:   no known conversion for argument 1 from ‘std::__shared_ptr_access<gtsam::PreconditionerParameters, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘gtsam::PreconditionerParameters’} to ‘const std::shared_ptr<gtsam::PreconditionerParameters>&’
  309 |       shared_ptr& operator=(const shared_ptr&) noexcept = default;
      |                             ^~~~~~~~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:313:2: note: candidate: ‘template<class _Yp> std::shared_ptr<_Tp>::_Assignable<const std::shared_ptr<_Yp>&> std::shared_ptr<_Tp>::operator=(const std::shared_ptr<_Yp>&) [with _Yp = _Yp; _Tp = gtsam::PreconditionerParameters]’
  313 |  operator=(const shared_ptr<_Yp>& __r) noexcept
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:313:2: note:   template argument deduction/substitution failed:
/home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:28709:26: note:   ‘std::__shared_ptr_access<gtsam::PreconditionerParameters, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘gtsam::PreconditionerParameters’} is not derived from ‘const std::shared_ptr<_Tp>’
28709 |   obj->preconditioner = *preconditioner;
      |                          ^~~~~~~~~~~~~~
In file included from /usr/include/c++/9/memory:81,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Testable.h:40,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Manifold.h:23,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Lie.h:26,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/VectorSpace.h:11,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/geometry/Point2.h:20,
                 from /home/chuan/Desktop/gtsam-develop/wrap/matlab.h:27,
                 from /home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/c++/9/bits/shared_ptr.h:324:2: note: candidate: ‘template<class _Yp> std::shared_ptr<_Tp>::_Assignable<std::auto_ptr<_Up> > std::shared_ptr<_Tp>::operator=(std::auto_ptr<_Up>&&) [with _Yp = _Yp; _Tp = gtsam::PreconditionerParameters]’
  324 |  operator=(auto_ptr<_Yp>&& __r)
      |  ^~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:324:2: note:   template argument deduction/substitution failed:
/home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:28709:26: note:   ‘std::__shared_ptr_access<gtsam::PreconditionerParameters, __gnu_cxx::_S_atomic, false, false>::element_type’ {aka ‘gtsam::PreconditionerParameters’} is not derived from ‘std::auto_ptr<T>’
28709 |   obj->preconditioner = *preconditioner;
      |                          ^~~~~~~~~~~~~~
In file included from /usr/include/c++/9/memory:81,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Testable.h:40,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Manifold.h:23,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/Lie.h:26,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/base/VectorSpace.h:11,
                 from /home/chuan/Desktop/gtsam-develop/gtsam/geometry/Point2.h:20,
                 from /home/chuan/Desktop/gtsam-develop/wrap/matlab.h:27,
                 from /home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:1:
/usr/include/c++/9/bits/shared_ptr.h:333:7: note: candidate: ‘std::shared_ptr<_Tp>& std::shared_ptr<_Tp>::operator=(std::shared_ptr<_Tp>&&) [with _Tp = gtsam::PreconditionerParameters]’
  333 |       operator=(shared_ptr&& __r) noexcept
      |       ^~~~~~~~
[... cut ...]
/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:33:53: error: incomplete type ‘Eigen::internal::traits<boost::serialization::U>’ used in nested name specifier
   33 |          has_write_access = (traits<Derived>::Flags & LvalueBit) ? 1 : 0,
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:81: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:27319: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2

deeplerming, 2024-10-25:
change   "obj->preconditioner = *preconditioner;" to
obj->preconditioner = std::make_shared<gtsam::PreconditionerParameters>(*preconditioner);

varunagrawal, 2024-10-25:
I made #1886 to fix this.

Huang-Chuan, 2024-10-26:
@varunagrawal I made changes as you did in #1886. But now I got new errors.

/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h: In instantiation of ‘struct Eigen::internal::accessors_level<boost::serialization::U>’:
/usr/include/boost/serialization/split_free.hpp:58:13:   required from ‘static void boost::serialization::free_loader<Archive, T>::invoke(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = Eigen::Matrix<double, -1, -1>]’
/usr/include/boost/serialization/split_free.hpp:74:18:   required from ‘void boost::serialization::split_free(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = Eigen::Matrix<double, -1, -1>]’
/home/chuan/Desktop/gtsam-develop/gtsam/base/MatrixSerialization.h:87:13:   required from ‘void boost::serialization::serialize(Archive&, gtsam::Matrix&, unsigned int) [with Archive = boost::archive::text_iarchive; gtsam::Matrix = Eigen::Matrix<double, -1, -1>]’
/usr/include/boost/serialization/serialization.hpp:126:14:   required from ‘void boost::serialization::serialize_adl(Archive&, T&, unsigned int) [with Archive = boost::archive::text_iarchive; T = Eigen::Matrix<double, -1, -1>]’
/usr/include/boost/archive/detail/iserializer.hpp:187:40:   [ skipping 30 instantiation contexts, use -ftemplate-backtrace-limit=0 to disable ]
/usr/include/boost/archive/detail/iserializer.hpp:461:22:   required from ‘static void boost::archive::detail::load_non_pointer_type<Archive>::invoke(Archive&, T&) [with T = gtsam::SO<-1>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/detail/iserializer.hpp:624:18:   required from ‘void boost::archive::load(Archive&, T&) [with Archive = boost::archive::text_iarchive; T = gtsam::SO<-1>]’
/usr/include/boost/archive/detail/common_iarchive.hpp:67:22:   required from ‘void boost::archive::detail::common_iarchive<Archive>::load_override(T&) [with T = gtsam::SO<-1>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/basic_text_iarchive.hpp:70:9:   required from ‘void boost::archive::basic_text_iarchive<Archive>::load_override(T&) [with T = gtsam::SO<-1>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/text_iarchive.hpp:82:52:   required from ‘void boost::archive::text_iarchive_impl<Archive>::load_override(T&) [with T = gtsam::SO<-1>; Archive = boost::archive::text_iarchive]’
/usr/include/boost/archive/detail/interface_iarchive.hpp:68:9:   required from ‘Archive& boost::archive::detail::interface_iarchive<Archive>::operator>>(T&) [with T = gtsam::SO<-1>; Archive = boost::archive::text_iarchive]’
/home/chuan/Desktop/gtsam-develop/build/wrap/gtsam/gtsam_wrapper.cpp:15594:18:   required from here
/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:32:54: error: incomplete type ‘Eigen::internal::traits<boost::serialization::U>’ used in nested name specifier
   32 |   enum { has_direct_access = (traits<Derived>::Flags & DirectAccessBit) ? 1 : 0,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/usr/include/eigen3/Eigen/src/Core/util/ForwardDeclarations.h:33:53: error: incomplete type ‘Eigen::internal::traits<boost::serialization::U>’ used in nested name specifier
   33 |          has_write_access = (traits<Derived>::Flags & LvalueBit) ? 1 : 0,
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
make[2]: *** [matlab/CMakeFiles/gtsam_matlab_wrapper.dir/build.make:81: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/__/wrap/gtsam/gtsam_wrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:27319: matlab/CMakeFiles/gtsam_matlab_wrapper.dir/all] Error 2

varunagrawal, 2024-10-26:
This is an issue with your Eigen installation. I cannot help you with this.

closed 2024-10-26 by varunagrawal