borglab/gtsam #1874: Running the example on windows

Oussamaaat, 2024-10-14:
## Description

## Steps to reproduce

1. run `cl /EHsc CameraResectioning.cpp` in the examples folder in powershell
2. Here's my folder tree : 
├───.github
│   ├───ISSUE_TEMPLATE
│   ├───scripts
│   └───workflows
├───.settings
├───build
│   ├───bin
│   │   └───Release
│   ├───cmake
│   │   └───CMakeFiles
│   ├───CMakeFiles
│   │   ├───009b702aa365ce99b45c4ddf405d7827
│   │   ├───024b76bdf713882b75b3fbd1a0ac6755
│   │   ├───0318edbe23eb9be893232c475b9d031d
│   │   ├───0bac22d37e19ed1def93b939d6984b0c
│   │   ├───0f1a6388d174a2737ce1777adcc4c812
│   │   ├───224f33dc68691e7e2892d0d711a4e6a1
│   │   ├───2ee9cc0e533aca9c076e698fc8cd519f
│   │   ├───3.30.4
│   │   │   ├───CompilerIdC
│   │   │   │   ├───Debug
│   │   │   │   │   └───CompilerIdC.tlog
│   │   │   │   └───tmp
│   │   │   ├───CompilerIdCXX
│   │   │   │   ├───Debug
│   │   │   │   │   └───CompilerIdCXX.tlog
│   │   │   │   └───tmp
│   │   │   ├───VCTargetsPath
│   │   │   │   └───x64
│   │   │   │       └───Debug
│   │   │   │           └───VCTargetsPath.tlog
│   │   │   └───x64
│   │   │       └───Debug
│   │   ├───3458a39990314220d236772c77c8b14a
│   │   ├───378e7d2a2a0002cc1956237e531f5d4d
│   │   ├───3b40b601a6ef63074ddcec0d50f482aa
│   │   ├───411887702e276b175b3f646ce73db9e2
│   │   ├───4834f0a8343ebc9701bec404e2dab4b0
│   │   ├───4fedd56015b94a799d2e3050a031cfb9
│   │   ├───58cb2c47234a55253b864b2321897c67
│   │   ├───5db79b4dc3d35f45ecbfd6b73126aba8
│   │   ├───6535872647f1f80b69f28c0a67b8d125
│   │   ├───77cc8ca66815340ab4607042eca546b9
│   │   ├───79331cbb0d944a09cc64a29f5e9faf48
│   │   ├───8243d3359e0764e0da1cc8ad86087179
│   │   ├───89021a349c2b3b00e0bbad3f3fbdd0e3
│   │   ├───8f524d56e3aa755b48f561bd71f97b77
│   │   ├───94385e383192518afea48c382ec6b26b
│   │   ├───9705e90498c2d5c227f32b64e8891ca9
│   │   ├───a111cf6ff36a58725e7e4206a8ce5945
│   │   ├───a26b24a6f0031982ae4c68858f332dfc
│   │   ├───a6254b280b03b544efc4207e7042b7cb
│   │   ├───ad5407f19d05aea1be214cc646f30654
│   │   ├───b68671162ad3cddeed9d1f9bc73236a6
│   │   ├───b7817b0e3976ff1aded27b9bd7cf3f97
│   │   ├───bbb9c42ab1b6fd35c948e949272c0c45
│   │   ├───bca33e3992f260536cfdae96c55cd539
│   │   ├───bceae929fcd7ccb9ee005c8d987bac47
│   │   ├───be0ad73519467a8133eea51c55ae7941
│   │   ├───c69bdef0acb1a3ce023c65ee659e02e6
│   │   ├───c7b8bd3573213bfdb5c459ea8199d905
│   │   ├───cefbada5b6e7bf88f43f681efa1b7372
│   │   ├───CMakeScratch
│   │   ├───d5b758e54895c3d0b306a85178c744c9
[... 496 lines cut ...]
    │   └───tools
    ├───scripts
    ├───sphinx
    ├───templates
    ├───tests
    │   ├───actual
    │   ├───expected
    │   │   ├───matlab
    │   │   │   ├───+gtsam
    │   │   │   │   ├───+MCU
    │   │   │   │   └───+OptimizerGaussNewtonParams
    │   │   │   ├───+ns1
    │   │   │   ├───+ns2
    │   │   │   │   └───+ns3
    │   │   │   └───+Pet
    │   │   ├───python
    │   │   └───xml
    │   └───fixtures
    └───utilities

## Expected behavior

## Environment

Thanks in advance !

varunagrawal, 2024-10-14:
I am not familiar with `cl`. Any reason you are not using CMake and then building the examples via the documented process?
It seems like in your process, the linker doesn't know the path to the GTSAM files for compilation.

Oussamaaat, 2024-10-14:
Thank you for your answer !

My goal is to know if i compiled succesfully and enabling matlab wrapper, but i'm not familiar with building c++ apps tried using the steps provided with the visual studio GUI but i failed.. so i used the following steps which seems to build without any errors but i have to check :

I installed boost cmake tbb and mkl

then : 

```
git clone https://github.com/borglab/gtsam.git
cd gtsam
mkdir build
cd build 

cmake .. -DCMAKE_BUILD_TYPE=Release -DGTSAM_BUILD_UNSTABLE:OPTION=OFF -DGTSAM_BUILD_EXAMPLES_ALWAYS:OPTION=OFF -DGTSAM_INSTALL_MATLAB_TOOLBOX:OPTION=ON -G "Visual Studio 17 2022" -A x64

cmake --build . --config Release
cmake --install . --config Release 

```

Then the command i mentioned before to run the example

Thank you for your time

varunagrawal, 2024-10-14:
After the above steps, you should be able to run
```sh
cl examples/CameraResectioning
```
which is generated as an executable in the build directory.

varunagrawal, 2024-10-14:
Note: You may have to add the correct extension, e.g. `cl examples/CameraResectioning.exe`.

Oussamaaat, 2024-10-15:
I don't know if it is related but i found out that several dependencies were not found on my machine when trying to use gtsam_wrapper.mexw64 with matlab : 

[image]