borglab/gtsam #1880: Python segmentation fault on noiseModel creation

aneesas, 2024-10-18:
## Description

I'm trying to run [this example](https://www.roboticsbook.org/S64_driving_perception.html#optimization-with-gtsam) from the Robotics book, and I'm getting a `Segmentation fault` on the second line creating the noise model. I get the same when I try to run the Python tests in GTSAM 4.2.0 (via `ninja python-test`). The C++ examples seem to run successfully (I tried `Pose2SLAMExample` and `Pose3Localization`). The segfault on noise model creation also occurs when I install GTSAM via pip.

## Steps to reproduce

Minimal failing code example:

```
import gtsam
gtsam.noiseModel.Diagonal.Sigmas((0.3, 0.3, 0.1))
```

or

```
import gtsam
import numpy as np
gtsam.noiseModel.Diagonal.Sigmas(np.array([0.3, 0.3, 0.1]))
```

## Expected behavior

I would expect the noise model to be successfully created; this is the way the examples seem to do it.

## Environment

Python 3.10.12
GTSAM 4.2.0 (occurs when installing via pip and also when building from source)
numpy 2.1.2
Ubuntu 22.04

## Additional information

From scouring other past issues, I've already tried enabling/disabling `MARCH_NATIVE` when building from source (no change). I thought [this](https://github.com/borglab/gtsam/issues/1720) issue might be relevant to what I'm seeing, but the `pybind11` variables in my `CMakeCache.txt` point to the wrap-included pybind version. I'm kind of stumped on what else to try.

diplodocuslongus, 2024-10-23:
I observe the same, with the same setup as yours, I actually commented on the issue you referred to just before noticing yours.
If you want to avoid the segfault, you will have to uninstall your numpy, matplotlib, and use the following combination:

- numpy 1.22.1
- matplpotlib 3.8

ProfFan, 2024-10-23:
We may need a new minor release to fix this, as this is most likely due to `numpy` 2.0 changes

aneesas, 2024-10-23:
Thanks for the tip! Yes, it looks like `numpy` 2.0 is the breaking version--I rolled the version back and it works with 1.26.4 but segfaults with 2.0.0.

ProfFan, 2024-10-27:
It is fixed in `develop` but the old release version is susceptible to this

varunagrawal, 2024-11-06:
Closing since we have a workaround.

closed 2024-11-06 by varunagrawal