borglab/gtsam #1908: Missing <cassert> include?

themightyoarfish, 2024-11-29:
## Description

I am experiencing a compile error when building gtsam from CMake, but not when doing it manually. Thus I am not sure when this gets triggered, but either way it could be a missing include.

## Steps to reproduce

1.  Create a CMakeLists file like this:

```
project(PointCloudIntegrator LANGUAGES CXX)

include(ExternalProject)

# Set the URL and desired GTSAM version/tag
set(GTSAM_REPO "https://github.com/borglab/gtsam.git")
set(GTSAM_VERSION "4.2a9")

# Set the installation directory for GTSAM
set(GTSAM_INSTALL_DIR "${CMAKE_BINARY_DIR}/gtsam_install")

# Use ExternalProject to download and build GTSAM
ExternalProject_Add(
    gtsam
    PREFIX ${CMAKE_BINARY_DIR}/gtsam
    GIT_REPOSITORY ${GTSAM_REPO}
    GIT_TAG ${GTSAM_VERSION}
    GIT_PROGRESS ON
    CMAKE_ARGS
        -DCMAKE_INSTALL_PREFIX=${GTSAM_INSTALL_DIR}
        -DGTSAM_BUILD_EXAMPLES_ALWAYS=OFF
        -DGTSAM_BUILD_TESTS=OFF
        -DGTSAM_WITH_TBB=OFF
        -DGTSAM_BUILD_WITH_MARCH_NATIVE=ON
        -DGTSAM_USE_SYSTEM_EIGEN=ON
        -DCMAKE_CXX_STANDARD=17
        -DCMAKE_VERBOSE_BUILD=ON
        -DCMAKE_VERBOSE_MAKEFILE=ON
)
```
2. `mkdir build && cd build`
3. `cmake .. && make`

This results in

```
<project>/build/gtsam/src/gtsam/gtsam/base/Vector.cpp:20:
<project>build/gtsam/src/gtsam/gtsam/base/Vector.h:224:3: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  224 |   assert (y.size()==x.size());
      |   ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
```

Meanwhile the same cmake args to a plain manual build do not cause this issue.

## Expected behavior

No compile error.

## Environment

macOS 15.1.1, cmake version 3.31.0, apple clang 16.0.0.

closed 2024-12-02 by ProfFan

ProfFan, 2024-12-03:
@themightyoarfish Could you check if #1910 fixes the issue?

themightyoarfish, 2024-12-04:
The error changes, but I think there's more offenders:

```
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/Vector.cpp:200:3: error: use of undeclared identifier 'assert'
  200 |   assert (b.size()==a.size());
      |   ^
1 error generated.
make[5]: *** [gtsam/CMakeFiles/gtsam.dir/base/Vector.cpp.o] Error 1
make[5]: *** Waiting for unfinished jobs....
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:136:9: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  136 |         assert(actualEndBlock < (DenseIndex)variableColOffsets_.size());
      |         ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:149:9: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  149 |         assert(actualEndBlock < (DenseIndex)variableColOffsets_.size());
      |         ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:195:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  195 |       assert(matrix_.cols() == variableColOffsets_.back());
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:196:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  196 |       assert(blockStart_ < (DenseIndex)variableColOffsets_.size());
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:197:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  197 |       assert(rowStart_ <= matrix_.rows());
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:198:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  198 |       assert(rowEnd_ <= matrix_.rows());
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:199:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  199 |       assert(rowStart_ <= rowEnd_);
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:204:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  204 |       assert(matrix_.cols() == variableColOffsets_.back());
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:205:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
  205 |       assert(block < (DenseIndex)variableColOffsets_.size() - 1);
      |       ^
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/base/VerticalBlockMatrix.h:206:7: error: use of undeclared identifier 'assert'; did you mean 'mpl_::assert_'?
[... cut ...]
/opt/homebrew/include/boost/mpl/assert.hpp:97:8: note: 'mpl_::assert_' declared here
   97 | struct assert_
      |        ^
10 errors generated.
make[5]: *** [gtsam/CMakeFiles/gtsam.dir/base/VerticalBlockMatrix.cpp.o] Error 1
In file included from /Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/basis/Chebyshev2.cpp:19:
/Users/rasmus/Documents/Projects/pointcloud-utils/build/gtsam/src/gtsam/gtsam/basis/Chebyshev2.h:65:5: error: use of undeclared identifier 'assert'
```

PR #1910 (2024-12-02): Remove the asserts as they have already been checked in Eigen
Fixes #1908 #1902 

Rationale: the dimensions are already both statically and dynamically checked in the templated Eigen methods.