borglab/gtsam #1920: Wrong `abs` used in `TranslationFactor`

ProfFan, 2024-12-07:
```cxx
/home/fan/src/gtsam/gtsam/sfm/TranslationFactor.h:137:30: warning: using integer absolute value function 'abs' when argument is of floating point type [-Wabsolute-value]
  137 |     const double abs_scale = abs(scale[0]);
      |                              ^
/home/fan/src/gtsam/gtsam/sfm/TranslationFactor.h:137:30: note: use function 'std::abs' instead
  137 |     const double abs_scale = abs(scale[0]);
      |                              ^~~
      |                              std::abs
```

varunagrawal, 2024-12-13:
Is this a false negative? `abs` is overloaded and unless we're using the C version, I guess the main thing is to just add the std::

dellaert, 2025-01-10:
Fixed in #1965

closed 2025-01-10 by dellaert