borglab/gtsam #1931: issue with copying gtsam objects in python

fatrybl, 2024-12-15:
Hello, there is a problem with copying items in Python 3.11 and GTSAM API. 
Some objects can be copied and deepcopied successfully, some others - no. 

For example, gtsam.NonlinearFactorGraph() instance can be successfully copied.
[image]
[image]

whereas copying gtsam.gtsam.PreintegratedCombinedMeasurements() instance  leads to errors in deepcopy module of Python.

[image]
[image]

as well as gtsam.gtsam.PreintegratedImuMeasurements (but another error).

[image]
[image]

this is clean python 3.11 and gtsam  4.2 installed with pip.

P.S.
In this [issue](https://github.com/borglab/gtsam/issues/1784) i left comments right after it had been closed. So maybe you could also take a look at it.

ProfFan, 2024-12-16:
You just need to edit `navigation.i`:

```cxx

class PreintegratedCombinedMeasurements {
// Constructors
  PreintegratedCombinedMeasurements(const gtsam::PreintegrationCombinedParams* params);
  PreintegratedCombinedMeasurements(const gtsam::PreintegrationCombinedParams* params,
				    const gtsam::imuBias::ConstantBias& bias);
  // Testable
  void print(string s = "Preintegrated Measurements:") const;
  bool equals(const gtsam::PreintegratedCombinedMeasurements& expected,
      double tol);

  // Standard Interface
  void integrateMeasurement(gtsam::Vector measuredAcc, gtsam::Vector measuredOmega,
      double deltaT);
  void resetIntegration();
  void resetIntegrationAndSetBias(const gtsam::imuBias::ConstantBias& biasHat);

  gtsam::Matrix preintMeasCov() const;
  double deltaTij() const;
  gtsam::Rot3 deltaRij() const;
  gtsam::Vector deltaPij() const;
  gtsam::Vector deltaVij() const;
  gtsam::imuBias::ConstantBias biasHat() const;
  gtsam::Vector biasHatVector() const;
  gtsam::NavState predict(const gtsam::NavState& state_i,
      const gtsam::imuBias::ConstantBias& bias) const;

  // enabling serialization functionality
  void serialize() const;
};
```

fatrybl, 2024-12-16:
@ProfFan,  the new line which is missing in the current implementation of navigation.i is : 
```
  // enabling serialization functionality
  void serialize() const;
```
right?
  then i copy-pasted  your full code  into gtsam/gtsam/navigation/navigation.i and freshly built with:
1. cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.11.10 -DCMAKE_INSTALL_PREFIX="./install" 
2. make check (tests passed)
3. make python-install
4. make python-tests (tests passed)
5. run my example with PreintegratedCombinedMeasurements from above and now it throws this error:

[image]

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  pybind11::detail::instance::get_value_and_holder: `gtsam.gtsam.PreintegrationParams' is not a pybind11 base of the given `gtsam.gtsam.PreintegrationCombinedParams' instance

Process finished with exit code 134 (interrupted by signal 6:SIGABRT)
```

However, running the example with gtsam.PreintegratedImuMeasurements works fine now (without calling NonlinearFactroGraph() before). So the problem still exists only for Combined imu measurements instance.

P.s.
i`m not a c++ developer so can`t fix it myself. i can only test the proposed solutions and fix python code.

fatrybl, 2024-12-16:
and few more examples:

Params and pim objects can be successfully copied while ImuFactor() can not.

[image]
[image]

as well as graph with ImuFactor can not be copied (but with different error) :

[image]
[image]

varunagrawal, 2024-12-16:
@fatrybl 2 things:

1. You are declaring the params incorrectly. They accept a 3d vector aka `np.asarray([0, 0, -9.81])`.
2. Please copy the code here rather than use screenshots? It makes it easier to try and reproduce the issue.

fatrybl, 2024-12-17:
@varunagrawal, I built the project using the changes to navigation.i from this [PR](https://github.com/borglab/gtsam/pull/1939)
and ran my examples:
1) Now i`m able to deepcopy pim and params (as in your unit tests in the PR).
2) Still not able to copy ImuFactor or CombinedImuFactor --> can not copy the graph with these factors. The errors are the same as in screenshots above.

However, copying other factors (i.e. PriorFactorPose3) and the graph with them works well:

```
from copy import deepcopy
import gtsam.noiseModel
import numpy as np
from gtsam.symbol_shorthand import B, V, X

noise = gtsam.noiseModel.Isotropic.Sigma(6, 1)
prior = gtsam.PriorFactorPose3(X(0), gtsam.Pose3(), noise)

graph = gtsam.NonlinearFactorGraph()
graph.push_back(prior)

prior_copy = deepcopy(prior)
graph_copy = deepcopy(graph)
```

However, to reproduce copy error:
```
from copy import deepcopy
import gtsam.noiseModel
import numpy as np
from gtsam.symbol_shorthand import B, V, X

params = gtsam.PreintegrationParams(np.asarray([0, 0, -9.81]))
pim = gtsam.PreintegratedImuMeasurements(params)
imu_odom = gtsam.ImuFactor(X(0), V(0), X(1), V(1), B(0), pim)

imu_odom_copy = deepcopy(imu_odom)
```
```
Traceback (most recent call last):
  File "/home/mark/Desktop/PhD/ModuSLAM/phd/test.py", line 11, in <module>
    imu_odom_copy = deepcopy(imu_odom)
                    ^^^^^^^^^^^^^^^^^^
  File "/home/mark/miniconda3/envs/moduslam/lib/python3.11/copy.py", line 161, in deepcopy
    rv = reductor(4)
         ^^^^^^^^^^^
TypeError: cannot pickle 'gtsam.gtsam.ImuFactor' object
```

P.s.
I used lists ans tuples for gravity and other input params for gtsam-based objects (not np.array vectors) cause it worked fine all the times and did not influence computations. It seemed for me that gtsam is converting inputs to vectors itself. But Thank you for mentioning this, i`ll convert explicitly all inputs if it`s necessary requirement.

varunagrawal, 2024-12-17:
So it seems that the wrapper we use around Pybind11 doesn't support deep copy yet ([Pybind11 DeepCopy](https://pybind11.readthedocs.io/en/stable/advanced/classes.html#deepcopy-support)).

Unfortunately, I cannot give you a timeline on when this feature will be added at the moment.

varunagrawal, 2024-12-17:
Update: if the unit tests for #1939 pass, then the deep copy functionality should be good to go.

fatrybl, 2024-12-18:
using the latest commits from the fix-pim-serialization branch with python 3.11 gives me the same error as in your pipeline for the unit test: test_ImuFactor (test_Serialization.TestDeepCopy)

varunagrawal, 2024-12-21:
So I figured it out. The `PreintegratedImuMeasurements` object needs at least one `integrateMeasurement` call before it can be properly serialized. I have updated the test and will merge once CI passes.

closed 2024-12-21 by varunagrawal

peci1, 2025-03-14:
More stuff that cannot be copied: ISAM2, ISAM2Params, LevenbergMarquardtParams...

PR #1939 (2024-12-21): Add `serialize()` to various preintegration classes
Fixes #1931