borglab/gtsam #1935: std::out_of_range of FixedLagSmoother methods

HeXu1, 2024-12-16:
Hi,
I have test BatchFixedLagSmoother and IncrementalFixedLagSmoother method to fuse imu and lidar data with gtsam::CombinedImuFactor and addExpressionFactor factors.

I have built the latest gtsam from source they all have bug when they start fixedlagsmooth,
the BatchFixedLagSmoother method output is:
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at

and the IncrementalFixedLagSmoother method output is:
terminate called after throwing an instance of 'std::out_of_range'
  what():  invalid key

hope for your reply, thanks.

HeXu1, 2024-12-16:
and I don't know how to output debug information like #1092
[DEBUG] (2022-02-03 16:13:45.949) OptimizerGTSAM.cpp:101::optimize() Optimizing graph...
[DEBUG] (2022-02-03 16:13:45.949) OptimizerGTSAM.cpp:146::optimize() hasGPSPrior=false
[DEBUG] (2022-02-03 16:13:45.949) OptimizerGTSAM.cpp:164::optimize() fill poses to gtsam...

RomanStadlhuber, 2024-12-17:
I just want to chip in and say that I am experiencing the same issue.

Here is a [minmal example](https://gist.github.com/RomanStadlhuber/2e1d83cf6900ac2e44447ebccfb0c11c#file-smoother-cpp) to reprocude this issue.
While there are existing issues like #1092 where this problem seemingly occurs at random, I have consitently observed this problem to occur whenever I am adding timestamps that exceed the smoother lag.

While my example above was created by piecing together parts of the preintegration and FixedLagSmoother examples respectively, it might be worth mentioning that the [FixedLagSmoother example](https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/examples/FixedLagSmootherExample.cpp) constructs a minimal problem where factors are only added up to the lag time, leaving open the question if there is any further logic that needs to be called after the smoothers window is "full"?

Maybe the author of that example, @svwilliams knows more about that.

Kind regards.

keevindoherty, 2025-01-10:
I'm not sure about the original poster, but in the snippet shared by @RomanStadlhuber I think this may be simply a usage error. On [line 52](https://gist.github.com/RomanStadlhuber/2e1d83cf6900ac2e44447ebccfb0c11c#file-smoother-cpp-L52) and again on [line 70](https://gist.github.com/RomanStadlhuber/2e1d83cf6900ac2e44447ebccfb0c11c#file-smoother-cpp-L70), you have:
```c++
    nodeTimestamps[nodeCount] = tCurr;
```

However, `nodeTimestamps` which is the `KeyTimestampMap` passed to the smoother is a map from _keys_ to timestamps. `nodeCount` in that gist is being used as the _index_ of several symbols (`X`, `V`, and `B`), but it is not the key representation of those symbols. To get the key, you want something like:
```c++
gtsam::Symbol x_n = X(nodeCount);
nodeTimestamps[x_n] = tCurr;
/// ...and so on for V and B
```
I _think_ the implicit cast from `Symbol` to `Key` should "just work." But if it doesn't, you can always do something like `x_n.key()` or `gtsam::Key(x_n)`. 

I believe what was happening in the gist was the following: since `nodeCount` is an `int`, C++ was happy to implicitly convert it to a `gtsam::Key`, which is itself really just a typedef for a `uint64_t`. Therefore, when trying to look up the keys for the actual variables in your factor graph in the `nodeTimestamps` map, those keys would not exist (and so the `out_of_range` error was exactly what one would expect).

HeXu1, 2025-01-10:
@keevindoherty thanks, I have tried as your recommend, but it does't work.

RomanStadlhuber, 2025-02-05:
@keevindoherty yes, many thanks!! I can confirm that was in fact it,
I have also had a look at the source code for Kimera-VIO and how they insert keys into the timestamp map, and it is done once for each value.

So instead what I had done previously, and which was incorrect

```c++
nodeTimestamps[nodeCount] = tCurr;
```

I did it like in Kimeras [VioBackend.cpp, Line 1163](https://github.com/MIT-SPARK/Kimera-VIO/blob/master/src/backend/VioBackend.cpp#L1163)

```c++
// insert predicted state into graph nodes
nodes.insert(X(nodeCount), predState.pose());
nodes.insert(V(nodeCount), predState.v());
nodes.insert(B(nodeCount), currBias);
// need one key for each value (that is pose, velocity and bias separately)
for (auto const &val : nodes)
{
    nodeTimestamps[val.key] = tCurr;
}
```

And now my minimal preintegration example runs through to the end.
@HeXu1 I have updated the code in [my gist](https://gist.github.com/RomanStadlhuber/2e1d83cf6900ac2e44447ebccfb0c11c#file-smoother-cpp). Can you confirm it works for you?

Thank you and kind regards.

HeXu1, 2025-02-06:
@RomanStadlhuber Yes,it works for your example,thanks a lot.

HeXu1, 2025-02-07:
The optimization time is significantly increased: I am fusing lidar and imu, and the optimized fixed lag is set to 0.5 seconds. With the iteration of optimization, the time consumption of state estimation is gradually increasing.
Have you encountered similar problems? Thanks.
@keevindoherty @RomanStadlhuber

HeXu1, 2025-02-07:
The IncrementalFixedLagSmoother mode has the situation where the time consumption increases, while the BatchFixedLagSmoother does not.

RomanStadlhuber, 2025-02-07:
@HeXu1 hard to tell where this might be coming from, but the `IncrementalFixedLagSmoother` has a ["debug logging" option](https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.cpp#L49) which you can activate using `SETDEBUG("IncrementalFixedLagSmoother update", true)` as in [this test](https://github.com/borglab/gtsam/blob/develop/gtsam_unstable/nonlinear/tests/testIncrementalFixedLagSmoother.cpp#L97).

Altenatively, you could play with the relinearization thresholds of iSAM2.

But I suggest building a release build with debug info and attaching a profiler to it to see where most of the computational load is coming from.

skohlbr, 2025-04-22:
FWIW I also see this slowdown when using the `IncrementalFixedLagSmoother` for (wheel) odometry and IMU fusion. The fusion process started out at ~10% CPU consumption and then after 3h took around 40% on a robot which prompted me to investigate a bit. So what apparently happens is that sometimes, there happen exactly as many re-eliminations as there are cliques and that's exactly when things take very long (because of full Bayes tree rebuild). The interesting part is that, as time progresses, the duration of this "taking long" increases. See examples below:

Some time into running through the dataset, the `update()` call defaults to < 0.3 ms. But when the full re-elimination happens, it jumps to 5.9 ms.
<pre>
 INFO|1745331746.869345908|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745331746.869592666|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 0.191 ms
 INFO|1745331746.869726067|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 79624) ---
  Reelimintated: 27  Relinearized: 6  Cliques: 378
 INFO|1745331746.869734863|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745331746.875724619|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 5.927 ms
 INFO|1745331746.876111556|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 79625) ---
  Reelimintated: 381  Relinearized: 0  Cliques: 381
 INFO|1745331746.876125359|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745331746.876415660|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 0.212 ms
 INFO|1745331746.876547958|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 79626) ---
  Reelimintated: 12  Relinearized: 0  Cliques: 378
</pre>

Further into the dataset, the default time stays at < 0.3 ms, but now the full re-elimination takes 20 ms (on a Core I7 machine, no less, so will be very noticable on weaker systems)
<pre>
 INFO|1745332013.999333253|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745332013.999577009|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 0.143 ms
 INFO|1745332013.999713901|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 272694) ---
  Reelimintated: 12  Relinearized: 6  Cliques: 378
 INFO|1745332013.999724849|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745332014.020104431|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 20.291 ms
 INFO|1745332014.020670593|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 272695) ---
  Reelimintated: 381  Relinearized: 0  Cliques: 381
 INFO|1745332014.020691185|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745332014.021080333|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 0.274 ms
 INFO|1745332014.021244426|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 272696) ---
  Reelimintated: 12  Relinearized: 0  Cliques: 378
</pre>

This happens also if `relinearizeSkip` is set to some very high number (i.e. it might happen with 20 iterations between, even if `relinearizeSkip` is set to 100), suggesting that it's not the relinearization, but likely the marginalization of old states where the time is spent. What mystifies me a bit is the fact that things become slower over time, while the problem size stays the same due to marginalization.

skohlbr, 2025-04-23:
@keevindoherty thanks for that hint! While I still see a slight duration increase over time of the `update()` when full re-elimination happens, the runtime is now more than halved from before. For the case of step 272695 in my dataset it for instance went from ~20ms to ~6.5 ms with otherwise all settings the same.

<pre>
 INFO|1745389909.747703484|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745389909.747915446|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 0.127 ms
 INFO|1745389909.748060888|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 272694) ---
  Reelimintated: 12  Relinearized: 6  Cliques: 378
 INFO|1745389909.748069772|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745389909.754656646|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 6.506 ms
 INFO|1745389909.755073380|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 272695) ---
  Reelimintated: 381  Relinearized: 9  Cliques: 381
 INFO|1745389909.755086059|/odom_fusion_offline_node|ros.odom_fusion: --- End ISAM2 Update Result ---
 INFO|1745389909.755405882|/odom_fusion_offline_node|ros.odom_fusion: Smoother update took: 0.223 ms
 INFO|1745389909.755551158|/odom_fusion_offline_node|ros.odom_fusion: --- ISAM2 Update Result (Step 272696) ---
  Reelimintated: 12  Relinearized: 6  Cliques: 378
</pre>