borglab/gtsam #1951: Ambiguous operator^ with new Eigen versions

mcm001, 2024-12-29:
## Problem

After bumping the [Eigen version we use](https://github.com/mcm001/gtsam/blob/develop/gtsam/3rdparty/Eigen/Eigen/src/plugins/CommonCwiseBinaryOps.inc#L130), I noticed that they also define an operator^ for component-wise bitwise xor. On windows, this results in this compiler output: https://github.com/PhotonVision/thirdparty-gtsam/actions/runs/12537998515/job/34962683956 

## Resolutions:
For my uses, I can either patch Eigen to remove that operator, or patch GTSAM to remove its operator^ overloads. The code that added them is 10+ years old at this point it looks like (https://github.com/borglab/gtsam/commit/a54d17720218ee6a471d18c7dc8d53e4c7e45d67 https://github.com/borglab/gtsam/commit/4297d24c968e477c2c9958aba2d256af38a4a976 https://github.com/borglab/gtsam/commit/73fe5b2be4e7dcf7ea3fb8d9ba0de005663f9b4a ), and some of these might be exposed to downstream users? But the patch set to remove them is super tiny (attached below).

dellaert, 2025-01-10:
Sorry, I missed this. I support just patching GTSAM. It is very unlikely that people are using this arguably not-so-smart operator overload outside the CG code inside GTSAM.

dellaert, 2025-01-10:
@mcm001 will you patch and PR?

mcm001, 2025-01-10:
Will do - thanks!

PR #1969 (2025-01-11): Remove operator^ from Matrix and SubgraphPreconditioner
Eigen recently added an operator^ for coefficient-wise bitwise xor, which conflicts with gtsam's operator^. Since gtsam's operator^ isn't xor, I opted to remove it.

Fixes #1951.
commit a0ae31f0 (2025-01-11): Remove operator^ from Matrix and SubgraphPreconditioner

Eigen recently added an operator^ for coefficient-wise bitwise xor,
which conflicts with gtsam's operator^. Since gtsam's operator^ isn't
xor, I opted to remove it.

Fixes #1951.
closed 2025-01-11 by dellaert