borglab/gtsam #1952: gtsam python gtsam_plot.plot_pose2d  patch.Ellipse.__init__() has changed.

PeterQFR, 2024-12-30:
## Description

The api for `matplotlib.patches.Ellipse().__init__()` appears to change. Currently using a version 3.8

## Steps to reproduce

1. Run examples/Pose2ISAM2Example.py   with numpy==1.26 and matplotlib==3.8
2. There iwll be an error 
```
Traceback (most recent call last):
  File "/home/peter-laptop/multiagent/venv/lib/python3.11/site-packages/gtsam/examples/Pose2ISAM2Example.py", line 178, in <module>
    Pose2SLAM_ISAM2_example()
  File "/home/peter-laptop/multiagent/venv/lib/python3.11/site-packages/gtsam/examples/Pose2ISAM2Example.py", line 164, in Pose2SLAM_ISAM2_example
    report_on_progress(graph, current_estimate, i)
  File "/home/peter-laptop/multiagent/venv/lib/python3.11/site-packages/gtsam/examples/Pose2ISAM2Example.py", line 42, in report_on_progress
    gtsam_plot.plot_pose2(0, current_estimate.atPose2(i), 0.5, marginals.marginalCovariance(i))
  File "/home/peter-laptop/multiagent/venv/lib/python3.11/site-packages/gtsam/utils/plot.py", line 279, in plot_pose2
    plot_pose2_on_axes(axes,
  File "/home/peter-laptop/multiagent/venv/lib/python3.11/site-packages/gtsam/utils/plot.py", line 252, in plot_pose2_on_axes
    plot_covariance_ellipse_2d(axes, origin, gPp)
  File "/home/peter-laptop/multiagent/venv/lib/python3.11/site-packages/gtsam/utils/plot.py", line 155, in plot_covariance_ellipse_2d
    e1 = patches.Ellipse(origin,
         ^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Ellipse.__init__() takes 4 positional arguments but 5 were given

```

## Expected behavior
No Error. 

This error can be fixed by replacing 
gtsam/utils/plot.py Line 156 with a kwarg for angle instead of existing positional arguments. It seems that matplotlib has changed angle to a kwarg. 
eg new line 156
```
e1 = patches.Ellipse(origin, 
                                   np.sqrt(w[0])*2*k,
                                   np.sqrt(W[1])*2*k,
                                   angle==np.rad2deg(angle),
                                   fill=False)
```

## Environment
linux, python,
matplotlib=3.8
numpy=1.26

## Additional information

Dream-t, 2025-01-03:
I also had the same issue, tried your method and it did work, thanks！

varunagrawal, 2025-01-05:
@PeterQFR are you using the PyPI version? We already fixed this in the latest `develop` branch. @truher has been working on a nightly python release build in #1906 which you may be interested in.

PeterQFR, 2025-01-05:
Hi @varunagrawal yes it was the PyPI version. I'll have a look at the nightly.
If you're already fixed this then we can close this issue.

varunagrawal, 2025-01-05:
Thanks for the quick response.

In case the nightly version isn't ready yet, you could always install from source. Takes a bit longer but you'll get lots of cool features including intellisense support in VS Code.

closed 2025-01-05 by varunagrawal

PeterQFR, 2025-01-05:
@varunagrawal, thanks this is my first time using the python version of gtsam. I have mainly been using the C++. It fills in a nice gap in the python ecosystem, for just providing an object-oriented handling of transformations. I am liking the fact that it is fully featured compared to the cpp libraries and will likely be my go-to for this.