borglab/gtsam #1967: ROS build farm are broken for develop branch

jlblancoc, 2025-01-10:
## Description

See: https://build.ros2.org/job/Jdev__gtsam__ubuntu_noble_amd64/136/

## Why?

Reason: the use of embedded Eigen version by default leads to not using the `SYSTEM` flag for those includes, so any warning there is treated as an error. 

## Solutions 
To modify CMake scripts so..
1) Use Eigen from the system version by default, 
2) Like (1), but only when it's detected we are building for the ROS build farm (kida hack, but have been there...)
3) Add the SYSTEM flag to embedded version of Eigen (if not done already, right? @Gold856  )

What do you think, @dellaert ?

I would like (1), but I know you like backwards compatible behavior, so the second choice would probably be (3), and if it's already done but is not working for some cmake limitation (cannot use SYSTEM for non-system directories? never checked that..), then use (2).

dellaert, 2025-01-10:
You are correctly modeling my view: (3) if possible, (2) if not :-)

jlblancoc, 2025-01-13:
SYSTEM was already there: 
https://github.com/borglab/gtsam/blob/develop/cmake/HandleEigen.cmake#L45

so it will have to be (2)...

Gold856, 2025-01-13:
What if we instead use a pragma in the problematic file to suppress that specific warning? That's probably a much cleaner solution and much more fine grained than using a pragma in an Eigen header and having it affect all downstream consumers.
Alternatively, if this is a big enough problem, we could put the pragma in DisableStupidWarnings.h like originally planned.

Gold856, 2025-01-13:
Yes, that's the header I was referencing. Although I still think using pragmas in the affected gtsam files is a better method, mostly because of the granularity of control we get and the fact that it means we don't need to mess with Eigen. Also, it's likely that system Eigen packages would also be affected, so this solution would ensure that the warnings don't trigger on system Eigen packages as well.

jlblancoc, 2025-01-13:
I agree with you, @Gold856 . 

If you have time for a small PR , please tag me so I can check it. If not I'll add it to my list...
Log with the error: https://build.ros2.org/job/Jdev__gtsam__ubuntu_noble_amd64/136/console

Key part: -Werror=maybe-uninitialized
``` 
In file included from /tmp/ws/src/gtsam/gtsam/3rdparty/Eigen/Eigen/Core:294,
03:39:15                  from /tmp/ws/src/gtsam/gtsam/3rdparty/Eigen/Eigen/Dense:1,
03:39:15                  from /tmp/ws/src/gtsam/gtsam/base/OptionalJacobian.h:24,
03:39:15                  from /tmp/ws/src/gtsam/gtsam/base/Matrix.h:27,
03:39:15                  from /tmp/ws/src/gtsam/gtsam/geometry/FundamentalMatrix.cpp:8:
03:39:15 In copy constructor ‘Eigen::PlainObjectBase<Derived>::PlainObjectBase(const Eigen::PlainObjectBase<Derived>&) [with Derived = Eigen::Matrix<double, 3, 1>]’,
03:39:15     inlined from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>&) [with _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’ at /tmp/ws/src/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/Matrix.h:414:65,
03:39:15     inlined from ‘gtsam::FundamentalMatrix::FundamentalMatrix(const gtsam::Matrix3&)’ at /tmp/ws/src/gtsam/gtsam/geometry/FundamentalMatrix.cpp:43:47:
03:39:15 /tmp/ws/src/gtsam/gtsam/3rdparty/Eigen/Eigen/src/Core/PlainObjectBase.h:512:17: error: ‘*(double*)((char*)&svd + offsetof(Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>,Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::<unnamed>.Eigen::SVDBase<Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2> >::m_singularValues.Eigen::Matrix<double, 3, 1, 0, 3, 1>::<unnamed>.Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::m_storage.Eigen::DenseStorage<double, 3, 3, 1, 0>::m_data.Eigen::internal::plain_array<double, 3, 0, 0>::array[2]))’ may be used uninitialized [-Werror=maybe-uninitialized]
03:39:15   512 |       : Base(), m_storage(other.m_storage) { }
```

ProfFan, 2025-01-20:
```
In file included from /tmp/ws/src/gtsam/gtsam/base/Testable.h:36,
                from /tmp/ws/src/gtsam/gtsam/inference/Symbol.h:21,
                from /tmp/ws/src/gtsam/examples/CombinedImuFactorsExample.cpp:39:
/tmp/ws/src/gtsam/gtsam/base/concepts.h:20: error: "BOOST_CONCEPT_USAGE" redefined [-Werror]
  20 | #define BOOST_CONCEPT_USAGE(concept) void check##concept()
     |
In file included from /usr/include/boost/concept_check.hpp:31,
                from /usr/include/boost/range/concepts.hpp:19,
                from /usr/include/boost/range/size_type.hpp:20,
                from /usr/include/boost/range/size.hpp:21,
                from /usr/include/boost/range/functions.hpp:20,
                from /usr/include/boost/range/iterator_range_core.hpp:38,
                from /usr/include/boost/lexical_cast.hpp:30,
                from /usr/include/boost/program_options/value_semantic.hpp:14,
                from /usr/include/boost/program_options/options_description.hpp:13,
                from /usr/include/boost/program_options.hpp:15,
                from /tmp/ws/src/gtsam/examples/CombinedImuFactorsExample.cpp:36:
/usr/include/boost/concept/usage.hpp:28: note: this is the location of the previous definition
  28 | #   define BOOST_CONCEPT_USAGE(model)                                    \
     |
cc1plus: all warnings being treated as errors
gmake[2]: *** [examples/CMakeFiles/CombinedImuFactorsExample.dir/build.make:76: examples/CMakeFiles/CombinedImuFactorsExample.dir/CombinedImuFactorsExample.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:23370: examples/CMakeFiles/CombinedImuFactorsExample.dir/all] Error 2
gmake: *** [Makefile:166: all] Error 2
---
```

ProfFan, 2025-01-20:
We should kill `CombinedImuFactorsExample.cpp` from building when boost is not enabled

PR #1978 (2025-01-23): Fix GCC 13 warnings
Fixes #1967 and fixes #1981.
closed 2025-01-23 by dellaert