borglab/gtsam #1972: Error when installing python module from the sources

Mikhail2017, 2025-01-11:
## Description

make scripts fail to install built gtsam python modules, returning 

```
Traceback (most recent call last):
  File "/home/mikhail/dev/gtsam/build1/venv/bin/pybind11-stubgen", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/mikhail/dev/gtsam/build1/venv/lib/python3.12/site-packages/pybind11_stubgen/__init__.py", line 319, in main
    run(
  File "/home/mikhail/dev/gtsam/build1/venv/lib/python3.12/site-packages/pybind11_stubgen/__init__.py", line 358, in run
    QualifiedName.from_str(module_name), importlib.import_module(module_name)
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mikhail/anaconda3/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1324, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'gtsam'
```
, when running make python-install

## Steps to reproduce

1. Create and source virtual environment
2. Build and install according to https://github.com/borglab/gtsam/blob/develop/python/README.md

## Expected behavior

Successfull installation

## Environment

Ubuntu 22
Python 3.12.2

## Additional information

varunagrawal, 2025-01-26:
What version of GTSAM are you using? I believe this is an issue in an older commit but has been fixed since.

Mikhail2017, 2025-02-07:
Hi. 4.2a8

varunagrawal, 2025-02-07:
Please try the latest version. It should be fixed in that.

Mikhail2017, 2025-02-17:
Tried 4.3.0 - still the same issue

varunagrawal, 2025-03-01:
Can you please provide more details about your CMake variables? I can't reproduce this exactly without more information.

varunagrawal, 2025-03-01:
I tried building using another virtual environment and I still can't reproduce this.

Did you run `pip install -r python/dev_requirements.txt`?
Also, you should be seeing GTSAM version `4.3a0`. I don't think we have a 4.3.0 yet.

And please double check your python executable and version within CMake. It is possible CMake is using another python interpreter and not the one within your virtual environment.

lewisjiang, 2025-03-12:
Similar issue here with the lastest gtsam and python 3.11. The requirements for python are all met, and in `make` it meets no issue. In my case, it says `No module named 'gtsam_unstable'`, and the issue disappears if I disable unstable in cmake from the very beginning.

ProfFan, 2025-03-12:
It's very hard to debug without information about your setup. For example:

1. Are you using `pyenv` or any kind of virtual environment?
2. If you are, are you sure that the environment you used during `cmake` is the same as the one you used this time?

Also, could you post your `make` logs during which you found the problem? For example, with `VERBOSE=1`? Thank you very much!

lewisjiang, 2025-03-12:
@ProfFan Thank you for reaching out. Here are some details of my installation.
1. I'm on Ubuntu 20.04, with cmake=3.16.3, CMAKE_CXX_COMPILER_VERSION=9.4.0
2. I created a conda virtual environment with python 3.11, and activated it in a new terminal. 
3. In the terminal, I installed the requirements with `pip install -r python/dev_requirements.txt`.
4. Then I moved to the `build` dir and ran `cmake .. -DGTSAM_BUILD_PYTHON=1 -DGTSAM_PYTHON_VERSION=3.11 -DCMAKE_INSTALL_PREFIX=/home/user/.local/install_dir/gtsam4.2dev`. It found the corrent python interpreter in conda, and showed pybind11 v2.13.1. Overall it finished cleanly.
5. Next I ran `make -j10`, it took some time but no error messages.
6. Finally I ran `make python-install`, and here is the output:
```
[  0%] Built target pybind_wrap_gtsam_unstable
[ 15%] Built target metis-gtsam
[ 30%] Built target cephes-gtsam
[ 75%] Built target gtsam
[ 84%] Built target gtsam_unstable
[ 84%] Built target gtsam_unstable_header
[ 84%] Built target gtsam_unstable_py
[ 90%] Built target pybind_wrap_gtsam
[ 90%] Built target gtsam_header
[100%] Built target gtsam_py
Scanning dependencies of target python-unstable-stubs
Traceback (most recent call last):
  File "/home/user/anaconda3/envs/gt/bin/pybind11-stubgen", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/user/anaconda3/envs/gt/lib/python3.11/site-packages/pybind11_stubgen/__init__.py", line 319, in main
    run(
  File "/home/user/anaconda3/envs/gt/lib/python3.11/site-packages/pybind11_stubgen/__init__.py", line 358, in run
    QualifiedName.from_str(module_name), importlib.import_module(module_name)
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/anaconda3/envs/gt/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1140, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'gtsam_unstable'
make[3]: *** [python/CMakeFiles/python-unstable-stubs.dir/build.make:94: python/CMakeFiles/python-unstable-stubs] Error 1
make[2]: *** [CMakeFiles/Makefile2:36246: python/CMakeFiles/python-unstable-stubs.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:36110: python/CMakeFiles/python-install.dir/rule] Error 2
make: *** [Makefile:15173: python-install] Error 2

```

I also find that if I disable unstable in cmake, the last step will finish cleanly. But the gtsam python library would be installed to `~/.local/lib/python3.11` instead of anaconda3's directory (same behaviour w/o DCMAKE_INSTALL_PREFIX). The venv in conda can still find the python package, but when I remove the venv with `conda env remove`, the gtsam package is left orphaned in my local install directory. I used to try release 4.2 with the same environment settings, it finished smoothly during compiling and was installed to the correct location, but when I import gtsam in python and create a `gtsam.Pose3()` from rotation and translation, it will encounter segmentation fault and exit.

ProfFan, 2025-03-18:
I am pretty sure your Shell setup has issues @lewisjiang . Please try the following:
1. In your shell, without using Anaconda, try build the Python extension and see if it works.
2. If that works, please DELETE the entire build directory, run CMake with whatever arguments you use, and try building and install
3. If that does not work, paste the CMakeCache.txt here.

lewisjiang, 2025-03-18:
@ProfFan After poking around I find that commenting `source /opt/ros/noetic/setup.bash` (generated by ROS) in .bashrc will lead to a successful build and install of the most recent commit of GTSAM. The library can be used successfully in conda's python. However, the gtsam library is still installed to `~/.local/lib/python3.11` instead of anaconda's directory (same behaviour w/o DCMAKE_INSTALL_PREFIX), which will be left orphaned if I remove the conda environment.

lewisjiang, 2025-03-18:
And the `CMakeCache.txt` files from system python and conda's python only differ in the version names of the related pythons, for example `.cpython-38-x86_64-linux-gnu.so` vs `.cpython-311-x86_64-linux-gnu.so`, `/usr/lib/x86_64-linux-gnu/libpython3.8.so` vs `/home/user/anaconda3/envs/gt/lib/libpython3.11.so`. Eventually one is them is installed to `~/.local/lib/python3.8/site-packages/` and the other to `~/.local/lib/python3.11/site-packages/`

varunagrawal, 2025-04-03:
Seems like the issue was identified.

For future reference, it might be a good idea to explicitly specify the python executable via
```
cmake -DPYTHON_EXECUTABLE=<path to python interpreter> ..
```

This ensures all the right libraries and python dependencies are loaded.

closed 2025-04-03 by varunagrawal