borglab/gtsam #1976: IncrementalFixedLagSmoother with Smart Factors: Marginalization Causes VariableIndex Assertion Failure

JD-Florez, 2025-01-14:
## Description

When using `IncrementalFixedLagSmoother` with `SmartProjectionPoseFactor<Cal3_S2>`, marginalizing out a pose included in a smart factor causes the following assertion failure:

`gtsam/gtsam/inference/VariableIndex.h:188: gtsam::FactorIndices& gtsam::VariableIndex::internalAt(gtsam::Key): Assertion` `item != index_.end()' failed.`

## Steps to reproduce
I created a unit test for a center-pointing agent, observing a cluster of 5 points:

```
/* ----------------------------------------------------------------------------

 * GTSAM Copyright 2010, Georgia Tech Research Corporation,
 * Atlanta, Georgia 30332-0415
 * All Rights Reserved
 * Authors: Frank Dellaert, et al. (see THANKS for the full author list)

 * See LICENSE for the license information

 * -------------------------------------------------------------------------- */

/**
 * @file    testIncrementalFixedLagSmootherSmart.cpp
 * @brief   Unit test for the Incremental Fixed-Lag Smoother with Smart Factors
 * @author  Juan-Diego Florez (jdfc3@gatech.edu)
 * @date    Jan 12, 2025
 */

#include <CppUnitLite/TestHarness.h>
#include <gtsam/base/debug.h>
#include <gtsam/geometry/Point2.h>
#include <gtsam/inference/Key.h>
#include <gtsam/inference/Ordering.h>
#include <gtsam/inference/Symbol.h>
#include <gtsam/linear/GaussianBayesNet.h>
#include <gtsam/linear/GaussianFactorGraph.h>
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>
#include <gtsam/slam/BetweenFactor.h>
#include <gtsam/slam/SmartProjectionPoseFactor.h>
#include <gtsam_unstable/nonlinear/IncrementalFixedLagSmoother.h>

using namespace std;
using namespace gtsam;

typedef SmartProjectionPoseFactor<Cal3_S2> SmartFactor;

// Create symbols for variables
Key PoseKey(size_t i) { return Symbol('x', i); }
Key PointKey(size_t j) { return Symbol('l', j); }

/**
 * Returns a Pose3 that places the camera on a center-pointing circular orbit
 * around `center`. `i` is the step index, and angleDeg is how many
 * degrees per step.
 */
Pose3 inspectionPose(size_t i, double angleDeg, double radius,
                     const Point3& center) {
  double angle = (angleDeg * M_PI / 180.0) * i;

  // Compute camera translation on a circle in X-Y around 'center.z'
  double camX = center.x() + radius * std::cos(angle);
  double camY = center.y() + radius * std::sin(angle);
  double camZ = 0.0;  // orbit in the plane z=0
  Point3 cameraPosition(camX, camY, camZ);

  Vector3 forward = (Vector3(center - cameraPosition)).normalized();
  Vector3 worldUp(0.0, 1.0, 0.0);
  Vector3 right = forward.cross(worldUp).normalized();
  Vector3 up = right.cross(forward).normalized();
[... 471 lines cut ...]
Landmark 4 in image coords:  696.44
621.369,
in cam coords:     1.8
1.11212
 35.044
IncrementalFixedLagSmoother::update() Start
Bayes Tree Before Update:
P( x0 x1 x2 )
END
Current Timestamp: 6
Marginalizable Keys: x0 
Constrained Keys: x0(0)  x1(1)  x2(1)  x3(1)  x4(1)  x5(1)  x6(1)  
Bayes Tree After Update, Before Marginalization:
P( x0 x1 x2 )
END
testIncrementalFixedLagSmootherSmart: /home/jdflo/packages/gtsam/gtsam/inference/VariableIndex.h:188: gtsam::FactorIndices& gtsam::VariableIndex::internalAt(gtsam::Key): Assertion `item != index_.end()' failed.
```

## Expected behavior

I expect that smart factors should be able to handle references to marginalized poses. If not, then the smart factor would need to be recreated without the marginalized pose.

## Environment

I am currently on gtsam commit d48b1fc8402e28d5b6af8f59cce8dbe87e66cf0c, on Ubuntu 22.04, and am using C++.

dellaert, 2025-01-20:
Thanks for assigning me, @varunagrawal ;-)
Unfortunately, I have not worked with smart factors in a while. @JD-Florez want to create a draft PR with what you think the solution could be? Or we could discuss it here before implementing.

dellaert, 2025-01-22:
Yes, I think your intuition is right. Except, there is a possible issue with 1, using the “current estimate”. We might want to use the Jacobian when the factor was *first* linearized, so called FEJ. See  https://pgeneva.com/downloads/papers/Chen2023IROS.pdf
That’s probably gravy though.

Step 2 is really two steps:
A. marginalization, already done in the fixed lag smoother
B. making a “1 pose smaller” factor, which has a baked in pose along with its Jacobian.

B involves some work (and unit tests) to even enable the smart factor to allow for that. That is not too hard, I think, but should be done in isolation and tested in a separate PR.

Another PR then, however, needs to somehow allow the incremental smoother to call this “fixPose” method on smart factors. 

It might be worth assuming that 
```
newFactor <- oldFactor.fixPose(pose)
```
Is already written and see how PR 2 could be done.

JD-Florez, 2025-01-23:
Alternatively, what if smart factors were made to operate with isam2's wildfire threshold, such that they can internally manage what measurements to consider for triangulation. Perhaps if a landmark is already settled/not changing by much, the measurements linked to old pose variables don't need to be considered. Since the fixed-lag smoother is based on isam2, this could be a more far-reaching functionality?

dellaert, 2025-01-23:
Hmmm. I think that enabling smart factors to fix poses with their marginalization values is more general, as that could be used with any scheme; fixed lag smoother, iSAM, batch with graph simplification etc.

JD-Florez, 2025-01-23:
So perhaps some implementation that first allows smart factors to internally marginalize poses, and then allows for selection? In an extended isam2 case, a smart factor could virtually cause a batch optimization over its pose history, no?

dellaert, 2025-01-24:
A smart factor by itself cannot marginalize a pose, because that pose might be connected to many smart factors. So I think what you need is a method 1st fixPose` at this value: it will then calculate the Jacobian and redefine the error function to add that additional cost.