borglab/gtsam #1981: Build fails on Ubuntu 24, with Eigen matrix uninitialized error.

Adam-Laidlaw, 2025-01-16:
Hello

I am attempting to build gtsam on Ubuntu 24 lts

Following the build commands:

git clone --recursive <repo_url>
cd gtsam
mkdir build
cd build
make

I get cmake output:
-- The CXX compiler identification is GNU 13.3.0
-- The C compiler identification is GNU 13.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE
-- Performing Test COMPILER_HAS_WSUGGEST_OVERRIDE - Success
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE
-- Performing Test COMPILER_HAS_WMISSING_OVERRIDE - Failed
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.83.0/BoostConfig.cmake (found suitable version "1.83.0", minimum required is "1.65") found components: serialization system filesystem thread program_options date_time timer chrono regex 
-- Found Eigen version: 3.4.0
CMake Deprecation Warning at gtsam/3rdparty/metis/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Looking for execinfo.h
-- Looking for execinfo.h - found
-- Looking for getline
-- Looking for getline - found
-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES) 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Building 3rdparty
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS) 
-- Building base
-- Building basis
-- Building constrained
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3a0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3a0
-- Install prefix: /usr/local
-- Wrote /home/adam/gtsam/build/GTSAMConfig.cmake
-- Wrote /home/adam/gtsam/build/GTSAM_UNSTABLEConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 13.3.0
--  CMake version                                    : 3.28.3
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/gmake
-- Build flags                                               
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Enabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            :  -O3 -DNDEBUG
--  Enable Boost serialization                       : ON
--  GTSAM_COMPILE_FEATURES_PUBLIC                    : cxx_std_17
--  GTSAM_COMPILE_OPTIONS_PUBLIC                     : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC                 : 
--  GTSAM_COMPILE_OPTIONS_PUBLIC_RELEASE             : 
--  GTSAM_COMPILE_DEFINITIONS_PUBLIC_RELEASE         : 
--  Use System Eigen                                 : false (Using version: 3.4.0)
--  Use System Metis                                 : OFF
--  Using Boost version                              : 1.83.0
--  Use Intel TBB                                    : Yes (Version: 2021.11.0)
--  Eigen will use MKL                               : MKL not found
--  Eigen will use MKL and OpenMP                    : OpenMP found but GTSAM_WITH_EIGEN_MKL is disabled
--  Default allocator                                : TBB
--  Cheirality exceptions enabled                    : YES
--  Build with ccache                                : Yes
-- Packaging flags
--  CPack Source Generator                           : TGZ
--  CPack Generator                                  : TGZ
-- GTSAM flags                                               
--  Quaternions as default Rot3                      : Disabled
--  Runtime consistency checking                     : Disabled
--  Build with Memory Sanitizer                      : Disabled
--  Rot3 retract is full ExpMap                      : Enabled
--  Pose3 retract is full ExpMap                     : Enabled
--  Enable branch merging in DecisionTree            : Enabled
--  Enable timing machinery                          : Disabled
--  Allow features deprecated in GTSAM 4.3           : Enabled
--  Metis-based Nested Dissection                    : Enabled
--  Use tangent-space preintegration                 : Enabled
-- MATLAB toolbox flags
--  Install MATLAB toolbox                           : Disabled
-- Python toolbox flags                                      
--  Build Python module with pybind                  : Disabled
-- ===============================================================
-- Configuring done (1.1s)
-- Generating done (1.1s)
-- Build files have been written to: /home/adam/gtsam/build

Then, after running make, he following error is received:

in file included from /usr/include/eigen3/Eigen/Core:294,
                 from /usr/include/eigen3/Eigen/Dense:1,
                 from /home/adam/gtsam/gtsam/base/OptionalJacobian.h:24,
                 from /home/adam/gtsam/gtsam/base/Matrix.h:27,
                 from /home/adam/gtsam/gtsam/geometry/FundamentalMatrix.cpp:8:
In copy constructor ‘Eigen::PlainObjectBase<Derived>::PlainObjectBase(const Eigen::PlainObjectBase<Derived>&) [with Derived = Eigen::Matrix<double, 3, 1>]’,
    inlined from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>&) [with _Scalar = double; int _Rows = 3; int _Cols = 1; int _Options = 0; int _MaxRows = 3; int _MaxCols = 1]’ at /usr/include/eigen3/Eigen/src/Core/Matrix.h:414:65,
    inlined from ‘gtsam::FundamentalMatrix::FundamentalMatrix(const gtsam::Matrix3&)’ at /home/adam/gtsam/gtsam/geometry/FundamentalMatrix.cpp:43:47:
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:512:17: error: ‘*(double*)((char*)&svd + offsetof(Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>,Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2>::<unnamed>.Eigen::SVDBase<Eigen::JacobiSVD<Eigen::Matrix<double, 3, 3, 0, 3, 3>, 2> >::m_singularValues.Eigen::Matrix<double, 3, 1, 0, 3, 1>::<unnamed>.Eigen::PlainObjectBase<Eigen::Matrix<double, 3, 1, 0, 3, 1> >::m_storage.Eigen::DenseStorage<double, 3, 3, 1, 0>::m_data.Eigen::internal::plain_array<double, 3, 0, 0>::array[2]))’ may be used uninitialized [-Werror=maybe-uninitialized]
  512 |       : Base(), m_storage(other.m_storage) { }
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/adam/gtsam/gtsam/geometry/FundamentalMatrix.cpp: In constructor ‘gtsam::FundamentalMatrix::FundamentalMatrix(const gtsam::Matrix3&)’:
/home/adam/gtsam/gtsam/geometry/FundamentalMatrix.cpp:38:29: note: ‘svd’ declared here
   38 |   Eigen::JacobiSVD<Matrix3> svd(F, Eigen::ComputeFullU | Eigen::ComputeFullV);
      |                             ^~~
cc1plus: all warnings being treated as errors

Two odd things:
-It's appearing to use my system eigen, even though it should not be
-My system eigen is also version 3.4, so should behave the same as the built in eigen.

How can I disable these 'warnings as errors' ?

Thanks for your help

mcm001, 2025-01-19:
Interesting - it looks like Eigen isn't being treated as a system include. This works in CI and locally for me though. Are you able to post the full compile command that's failing?

juanlm-verus, 2025-01-21:
I'm having a similar issue. You can pass a flag to temporarily ignore the warnings, but after doing so, I still run into another different make error. gtsam 4.2 is also building/installing just fine for me too.

```

cmake .. -DCMAKE_BUILD_TYPE=Release -DGTSAM_USE_SYSTEM_EIGEN=OFF -DGTSAM_WITH_EIGEN_UNSUPPORTED=ON -DCMAKE_CXX_FLAGS="-Wno-error=maybe-uninitialized -Wno-error=array-bounds" -DEIGEN_MAX_ALIGN_BYTES=0
-- GTSAM_POSE3_EXPMAP=ON, enabling GTSAM_ROT3_EXPMAP as well
-- Found Eigen version: 3.4.0
CMake Deprecation Warning at gtsam/3rdparty/metis/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- checking for thread-local storage - found
-- Could NOT find MKL (missing: MKL_INCLUDE_DIR MKL_LIBRARIES)
-- Building 3rdparty
-- Installing Eigen Unsupported modules
-- Could NOT find GeographicLib (missing: GeographicLib_LIBRARY_DIRS GeographicLib_LIBRARIES GeographicLib_INCLUDE_DIRS)
-- Building base
-- Building basis
-- Building constrained
-- Building geometry
-- Building inference
-- Building symbolic
-- Building discrete
-- Building hybrid
-- Building linear
-- Building nonlinear
-- Building sam
-- Building sfm
-- Building slam
-- Building navigation
-- GTSAM Version: 4.3a0
-- Install prefix: /usr/local
-- Building GTSAM - as a SHARED library
-- Building base_unstable
-- Building geometry_unstable
-- Building linear_unstable
-- Building discrete_unstable
-- Building dynamics_unstable
-- Building nonlinear_unstable
-- Building slam_unstable
-- Building partition_unstable
-- GTSAM_UNSTABLE Version: 4.3a0
-- Install prefix: /usr/local
-- Wrote /mnt/c/Users/juan.lopera/gtsam/build/GTSAMConfig.cmake
-- Wrote /mnt/c/Users/juan.lopera/gtsam/build/GTSAM_UNSTABLEConfig.cmake
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- ===============================================================
-- ================  Configuration Options  ======================
--  CMAKE_CXX_COMPILER_ID type                       : GNU
--  CMAKE_CXX_COMPILER_VERSION                       : 13.3.0
--  CMake version                                    : 3.28.3
--  CMake generator                                  : Unix Makefiles
--  CMake build tool                                 : /usr/bin/gmake
-- Build flags
--  Build Tests                                      : Enabled
--  Build examples with 'make all'                   : Enabled
--  Build timing scripts with 'make all'             : Disabled
--  Build shared GTSAM libraries                     : Enabled
--  Put build type in library name                   : Enabled
--  Build libgtsam_unstable                          : Enabled
--  Build GTSAM unstable Python                      : Enabled
--  Build MATLAB Toolbox for unstable                : Disabled
--  Build for native architecture                    : Disabled
--  Build type                                       : Release
--  C compilation flags                              :  -O3 -DNDEBUG
--  C++ compilation flags                            : -Wno-error=maybe-uninitialized -Wno-error=array-bounds -O3 -DNDEBUG
--  Enable Boost serialization                       : ON
[... 812 lines cut ...]
  128 |   std::chrono::nanoseconds durationLOST;
      |                            ^~~~~~~~~~~~
[ 80%] Linking CXX executable TriangulationLOSTExample
[ 80%] Built target TriangulationLOSTExample
[ 80%] Linking CXX executable StereoVOExample_large
[ 80%] Built target StereoVOExample_large
[ 80%] Linking CXX executable ShonanAveragingCLI
[ 80%] Built target ShonanAveragingCLI
[ 80%] Linking CXX executable SFMExampleExpressions
[ 80%] Built target SFMExampleExpressions
[ 80%] Linking CXX executable SolverComparer
[ 80%] Linking CXX executable SFMExampleExpressions_bal
[ 80%] Built target SFMExampleExpressions_bal
[ 80%] Built target SolverComparer
[ 80%] Linking CXX executable SFMExample_SmartFactorPCG
[ 80%] Linking CXX executable SFMExample_SmartFactor
[ 80%] Built target SFMExample_SmartFactorPCG
[ 80%] Built target SFMExample_SmartFactor
make: *** [Makefile:166: all] Error 2

```

I apologize if this is too much content or not the proper way of providing this information. Please let me know if there is further information you need to look into this.

Thank you.

mcm001, 2025-01-21:
Per conversation with @Gold856 on Discord, we should probably consider `-Wno-error=maybe-uninitialized -Wno-error=array-bounds` by default. He can repro on a WSL instance running Ubuntu 24.04 and GCC 13.

ProfFan, 2025-01-21:
Could you use `make`'s debug mode and post the list of includes in the `cc` invocations? My gut feeling is that system Eigen leaked via some include paths.

PR #1978 (2025-01-23): Fix GCC 13 warnings
Fixes #1967 and fixes #1981.
closed 2025-01-23 by dellaert