borglab/gtsam #2008: Example for adding landmarks incrementally

Smithangshu, 2025-01-28:
Hi GTSAM team,

Thank you for your time and effort to make such a wonderful library. Also, you have extensive range of examples. But I think novice like me can even take more advantage of it, if we have an example of ISAM2 that adds landmarks incrementally, all the existing examples are adding all the landmarks to the graph at the first pose, so it is really hard for novice like me to understand what are things that we need to do if we want to add more landmarks to the calculation when the system is already running. Because practically, it is inevitable that new landmark will get generated after the first frame. So, I will request please add an example for doing it with ISAM2 and GenericProjectionFactor. And before making the example ready, if you kindly provide me a clue about how to proceed in this situation, would be highly appreciated.

Thank you.

varunagrawal, 2025-01-28:
Hi, have you seen `VisualISAM2Example.cpp` and `ISAM2Example_SmartFactor.cpp`? They both deal with incremental inference and adding landmarks and poses over the trajectory of the robot.

closed 2025-01-29 by varunagrawal

Smithangshu, 2025-01-29:
Sorry, but I think there is some confusion. With landmark I am trying to mean the 3D points of the landmark not the 2D measurements of the 3D points. In `VisualISAM2Example.cpp`, 3D points are added. But in `ISAM2Example_SmartFactor.cpp` there is no question of having 3D points because, it only needs the corresponding 2d points from different poses.

I have checked both the example files from the `develop` branch. I usually denote Landmark for those which are denoted by symbol 'l'.

The following code from `VisualISAM2Example.cpp` clearly shows that Landmark or 3D position whatever you call it is getting added to initial estimate on the very first pose but not in other poses. Rest of the loops are just adding the measurements for the landmarks and the subsequent pose's initial estimate.

    // If this is the first iteration, add a prior on the first pose to set the
    // coordinate frame and a prior on the first landmark to set the scale Also,
    // as iSAM solves incrementally, we must wait until each is observed at
    // least twice before adding it to iSAM.
    if (i == 0) {
      // Add a prior on pose x0, 30cm std on x,y,z and 0.1 rad on roll,pitch,yaw
      static auto kPosePrior = noiseModel::Diagonal::Sigmas(
          (Vector(6) << Vector3::Constant(0.1), Vector3::Constant(0.3))
              .finished());
      graph.addPrior(Symbol('x', 0), poses[0], kPosePrior);

      // Add a prior on landmark l0
      static auto kPointPrior = noiseModel::Isotropic::Sigma(3, 0.1);
      graph.addPrior(Symbol('l', 0), points[0], kPointPrior);

      // Add initial guesses to all observed landmarks
      // Intentionally initialize the variables off from the ground truth
      static Point3 kDeltaPoint(-0.25, 0.20, 0.15);
      for (size_t j = 0; j < points.size(); ++j)
        initialEstimate.insert<Point3>(Symbol('l', j), points[j] + kDeltaPoint);

    }

And what you are referring as getting added in every pose as part of loop is the measurement of each of the landmarks. But this was not my topic for which I have tried to draw your attention. 

Hope now, it will make sense about what I am referring to.

Btw, I have placed a request to join the Google group regarding different matters apart from this one, several days ago but could not be part of the group yet.

varunagrawal, 2025-01-30:
Oh I see. Apologies for the confusion.

That should be a simple fix. Move this block
```cpp
  // Add initial guesses to all observed landmarks
  // Intentionally initialize the variables off from the ground truth
  static Point3 kDeltaPoint(-0.25, 0.20, 0.15);
  for (size_t j = 0; j < points.size(); ++j)
    initialEstimate.insert<Point3>(Symbol('l', j), points[j] + kDeltaPoint);
```
to the else block.

Ideally, the `if (i == 0)` block should be executed before the for loop.
Here's an updated script for you:

```cpp
/* ----------------------------------------------------------------------------

 * GTSAM Copyright 2010, Georgia Tech Research Corporation,
 * Atlanta, Georgia 30332-0415
 * All Rights Reserved
 * Authors: Frank Dellaert, et al. (see THANKS for the full author list)

 * See LICENSE for the license information

 * -------------------------------------------------------------------------- */

/**
 * @file    VisualISAM2Example.cpp
 * @brief   A visualSLAM example for the structure-from-motion problem on a
 * simulated dataset This version uses iSAM2 to solve the problem incrementally
 * @author  Duy-Nguyen Ta
 */

/**
 * A structure-from-motion example with landmarks
 *  - The landmarks form a 10 meter cube
 *  - The robot rotates around the landmarks, always facing towards the cube
 */

// For loading the data
#include "SFMdata.h"

// Camera observations of landmarks will be stored as Point2 (x, y).
#include <gtsam/geometry/Point2.h>

// Each variable in the system (poses and landmarks) must be identified with a
// unique key. We can either use simple integer keys (1, 2, 3, ...) or symbols
// (X1, X2, L1). Here we will use Symbols
#include <gtsam/inference/Symbol.h>

// We want to use iSAM2 to solve the structure-from-motion problem
// incrementally, so include iSAM2 here
#include <gtsam/nonlinear/ISAM2.h>

// iSAM2 requires as input a set of new factors to be added stored in a factor
// graph, and initial guesses for any new variables used in the added factors
#include <gtsam/nonlinear/NonlinearFactorGraph.h>
#include <gtsam/nonlinear/Values.h>

// In GTSAM, measurement functions are represented as 'factors'. Several common
// factors have been provided with the library for solving robotics/SLAM/Bundle
// Adjustment problems. Here we will use Projection factors to model the
// camera's landmark observations. Also, we will initialize the robot at some
// location using a Prior factor.
#include <gtsam/slam/ProjectionFactor.h>

#include <vector>

using namespace std;
[... 64 lines cut ...]
    // as iSAM solves incrementally, we must wait until each is observed at
    // least twice before adding it to iSAM.

    // Update iSAM with the new factors
    isam.update(graph, initialEstimate);
    // Each call to iSAM2 update(*) performs one iteration of the iterative
    // nonlinear solver. If accuracy is desired at the expense of time,
    // update(*) can be called additional times to perform multiple optimizer
    // iterations every step.
    isam.update();
    Values currentEstimate = isam.calculateEstimate();
    cout << "****************************************************" << endl;
    cout << "Frame " << i << ": " << endl;
    currentEstimate.print("Current estimate: ");

    // Clear the factor graph and values for the next iteration
    graph.resize(0);
    initialEstimate.clear();
  }

  return 0;
}
/* ************************************************************************* */

```

Smithangshu, 2025-02-07:
Hi Varun,

Sorry for my late reply. I was occupied with some other issues during these days. 

Thank you for understanding my point. But the example code does not align with what I am saying. For example, suppose this is a monocular scenario, here we don't have the privilege to add first 3d points as prior to the graph before we start processing or even adding constrains and initial estimates for all the point since the first frame. Rather we need to wait for some frames to process to get 3d landmarks generated then we can add the first landmark as prior to the graph, then eventually we can add the constrains and initial estimates that too in procedural manner instead of all at once. I know this is complex, but with such advance implementation example it will help to understand how to use GTSAM in such advance cases with it's all glory.

One more thing, I am having in my mind, is it possible to have inverse depth parameterization with ISAM2. I have checked `gtsam_unstable` and found 3 types of InvDepthFactors but could not found any example or clue about how to use them procedurally. Should I create raise issue for requesting example for this?

Smithangshu, 2025-02-07:
Basically, I am not asking question because I already know the answer, rather I am requesting you to add at least a few examples that would show how it can be used this way. Because all the examples we have are adding all the landmarks and landmark prior at very first pose, which easier for understand to those who are using GTSAM for stereo or lidar based system when 3d landmarks are readily available.

dellaert, 2025-02-08:
@Smithangshu you are welcome to contribute an example :-) Just create a branch on a fork and do a PR. If GTSAM is useful to you and you have figured out the answer, this is one way to contribute back.

Smithangshu, 2025-02-10:
Hello Professor @dellaert, 

I will surely try to contribute for this library. But only after I will complete what I am trying since last 7 years. It's taking long because I am just an engineering graduate in Information Technology, working as a software freelancer. I don't have any one to guide me to my goal. But I promise I will add my contribution to this library. 

Sir, can I mail you if I need to some guidance regarding things around this library?