borglab/gtsam #2037: Use Docker for CI

varunagrawal, 2025-03-01:
## Feature

Have a custom docker image uploaded to the [borglab dockerhub account](https://hub.docker.com/r/borglab/gtsam) which has all the dependencies pre-installed.
We can then use these images in the CI and simplify+speedup the overall process.

## Motivation

Currently, the CI pulls various dependencies and installs them. Using a docker image which already has these dependencies installed can provide a nice speedup of the CI.

## Pitch

This is not very difficult to do but requires some time and familiarity with respect to Docker, Github Actions, and how these two work together.

The impact of this feature would be quite large, since it would lead to better development turnaround time.

suveergarg, 2025-03-12:
I am interested in this and can take this up.

suveergarg, 2025-03-12:
Got it. Would the CI pipeline mostly be pulling docker images or also building and pushing them to the docker hub account?

suveergarg, 2025-03-21:
@varunagrawal  I created a MR as a starting point for the above : https://github.com/borglab/gtsam/pull/2065