borglab/gtsam #2050: Apt package build fails on Ubuntu 20.04, seems related to "Hat"

berndpfrommer, 2025-03-10:
Last night's build failed with the below error (and many more following).

I have trouble figuring out what the root cause of it is. It has something to do with Eigen, something to do with "Hat". Most likely the Eigen version (3.3.7.2) of Ubuntu 20.04 does not support some feature that is used with some of the most recent commits. Unfortunately there have been a number of commits the last few days  and I'm not sure which one is the culprit. 

```
cd /<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/gtsam && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_TIMER_DYN_LINK -DNDEBUG -Dgtsam_EXPORTS -I/<<PKGBUILDDIR>> -I/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu -I/<<PKGBUILDDIR>>/CppUnitLite -I/<<PKGBUILDDIR>>/gtsam/3rdparty/metis/libmetis -I/<<PKGBUILDDIR>>/gtsam/3rdparty/metis/GKlib -I/<<PKGBUILDDIR>>/gtsam/3rdparty/cephes -isystem /<<PKGBUILDDIR>>/gtsam/3rdparty/SuiteSparse_config -isystem /<<PKGBUILDDIR>>/gtsam/3rdparty/Spectra -isystem /<<PKGBUILDDIR>>/gtsam/3rdparty/CCOLAMD/Include -isystem /usr/include/eigen3  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -g -DNDEBUG -fPIC   -Werror -Wall -Wpedantic -Wextra -Wno-unused-parameter -Wreturn-local-addr -Wreturn-type -Wformat -Werror=format-security -Wsuggest-override -g -O3 -Wno-unused-local-typedefs -std=c++17 -o CMakeFiles/gtsam.dir/geometry/Line3.cpp.o -c /<<PKGBUILDDIR>>/gtsam/geometry/Line3.cpp
In file included from /<<PKGBUILDDIR>>/gtsam/geometry/CalibratedCamera.h:23,
                 from /<<PKGBUILDDIR>>/gtsam/geometry/CalibratedCamera.cpp:20:
/<<PKGBUILDDIR>>/gtsam/geometry/Pose3.h: In static member function ‘static gtsam::Pose3::LieAlgebra gtsam::Pose3::wedge(double, double, double, double, double, double)’:
/<<PKGBUILDDIR>>/gtsam/geometry/Pose3.h:407:52: error: no matching function for call to ‘Eigen::Matrix<double, 6, 1>::Matrix(double&, double&, double&, double&, double&, double&)’
  407 |     return Hat(TangentVector(wx, wy, wz, vx, vy, vz));

```

oursland, 2025-03-10:
Could possibly be related to https://github.com/borglab/gtsam/pull/2045

ProfFan, 2025-03-12:
#2053 should fix this, thank you all for the report!

closed 2025-03-12 by ProfFan