borglab/gtsam #2058: Unable to build GTSAM without Boost after latest update

Smithangshu, 2025-03-15:
## Description

With the latest version whenever I am trying to build the library without `BOOST` in `Windows` with `CMake`, it fails. But in previous version. I could build it without any issue.

I am getting following error

`3>D:\Gits\gtsam\gtsam\base\Lie.h(316,1): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
3>(compiling source file '../../../gtsam/gtsam/precompiled_header.cpp')
3>D:\Gits\gtsam\gtsam\base\Lie.h(316,1):`

## Steps to reproduce

1. Just disable all the Boost related flags in CMake GUI.
2. Try to compile the build with VS 2022.

## Expected behavior

It should work without any issue.

## Environment
Windows 11.
VS 2022.
CMake GUI.

## Additional information

ProfFan, 2025-03-15:
Could be #2047

dellaert, 2025-03-15:
@Smithangshu could you try with #2047 undone?

Smithangshu, 2025-03-16:
@ProfFan yes, it is the cause.

Smithangshu, 2025-03-16:
@dellaert if I comment out this section from Lie.h. And build it, then it works like charm.

[image]

dellaert, 2025-03-16:
Ah, that’s my bad! It should now use the new macro. It was my Hat/Vee that probably landed concurrently with @varunagrawal ‘s PR

ProfFan, 2025-03-17:
Should be solved with #2060

closed 2025-03-17 by ProfFan