borglab/gtsam #2089: Build fails on 32-bit architectures

dkogan, 2025-04-12:
Hello.

The Debian release of gtsam failed to rebuild on the 32-bit architectures: https://buildd.debian.org/status/package.php?p=gtsam

Click on "Build-Attempted" to get the build log. It looks like things build, but the tests fail.

Let me know if you need more info. Thanks.

ProfFan, 2025-04-13:
Seems to be a `typeinfo` related issue. Can't currently determine whether it's in Pybind11 or GTSAM itself.

Could you try current `develop` on the same build environment? Many thanks!

dkogan, 2025-04-17:
Hi. It's quite a bit of work to try to rebuild on `develop` because there are a lot of patches that were needed for the packaging work, and they will need to be updated to apply to `develop` successfully: https://salsa.debian.org/science-team/gtsam/-/tree/master/debian/patches?ref_type=heads

Getting as many of these as possible merged upstream would be good. Maybe yall already did that, and that's why the patches don't apply anymore.

Is trying with the latest `develop` branch required for you to work on this? I'm fully aware that you don't want to work on anything that's not your bleeding edge, but the new Debian release will be out in a few months, and 4.2 is what we'll ship. Let me know if you are able to debug 4.2. I can also run experiments myself, if you can suggest specific things.

Thanks

ProfFan, 2025-04-17:
Hi @dkogan, I was very busy (defended my thesis today) but now I have more time on this. I can run some more tests in the following week. By the way, what I mean is not building the debian package, but just building the `develop` version on the same environment as the build machine, including the python wrapper, and test if that works :)

dkogan, 2025-04-29:
Congrats on the defense! I tried to just build the develop branch on i386 without the debian stuff, and it's not entirely clear what that's doing. Any able to debug this, or can you suggest some things to me about how to debug? Thanks

ProfFan, 2025-05-01:
I'll try spin up an ARMv7 box to see what is wrong with it

ProfFan, 2025-05-01:
Just run the test suite on i386/ARM (I think ARM should have better tooling support nowadays) with `develop` (`make python-tests`), and see if that works. Also check if `make check` has any failures. If only Python is failing, most likely it's due to `typeinfo` comparisons as from the log in your Debian build it seems only type conversions are failing.

dkogan, 2025-05-06:
Hi. I just did this at this commit:

```
commit 42cc0ac922460d506a7bbf3a37a4a0b1e103b79e (HEAD -> develop, origin/develop, origin/HEAD)
Merge: beceb654b 8f5e78c2b
Author: Frank Dellaert <dellaert@users.noreply.github.com>
Date:   Sun Apr 27 17:56:43 2025 -0400

    Merge pull request #2111 from borglab/new-docs-slam
    
    New docs: SLAM
```

- made an i386 chroot: `mmdebstrap --arch=i386 sid root_sid_i386`
- added an `schroot` config for it
- `apt build-dep gtsam` to get the build-dependendies
- `mkdir build && cd build && cmake -DGTSAM_BUILD_UNSTABLE=OFF -DGTSAM_UNSTABLE_BUILD_PYTHON=OFF -DGTSAM_BUILD_PYTHON=ON ..`
- `make python-test`

This failed compiling `python/slam.cpp` runs out of memory. As a workaround, I made the contents of `slam()` empty. Then it was able to build. Then I see the same errors as the debian log shows. Suggestions to fix? If it is the `typeinfo` thing you suspect, then how do I resolve it?

ProfFan, 2025-05-07:
Let me find some time tomorrow to debug, but as a tip:
```
TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. gtsam.gtsam.JacobianFactor()
    2. gtsam.gtsam.JacobianFactor(b_in: numpy.ndarray[numpy.float64[m, 1]])
    3. gtsam.gtsam.JacobianFactor(i1: int, A1: numpy.ndarray[numpy.float64[m, n]], b: numpy.ndarray[numpy.float64[m, 1]], model: gtsam.gtsam.noiseModel.Diagonal)
    4. gtsam.gtsam.JacobianFactor(i1: int, A1: numpy.ndarray[numpy.float64[m, n]], i2: int, A2: numpy.ndarray[numpy.float64[m, n]], b: numpy.ndarray[numpy.float64[m, 1]], model: gtsam.gtsam.noiseModel.Diagonal)
    5. gtsam.gtsam.JacobianFactor(i1: int, A1: numpy.ndarray[numpy.float64[m, n]], i2: int, A2: numpy.ndarray[numpy.float64[m, n]], i3: int, A3: numpy.ndarray[numpy.float64[m, n]], b: numpy.ndarray[numpy.float64[m, 1]], model: gtsam.gtsam.noiseModel.Diagonal)
    6. gtsam.gtsam.JacobianFactor(graph: gtsam::GaussianFactorGraph)
    7. gtsam.gtsam.JacobianFactor(graph: gtsam::GaussianFactorGraph, p_variableSlots: gtsam::VariableSlots)
    8. gtsam.gtsam.JacobianFactor(graph: gtsam::GaussianFactorGraph, ordering: gtsam.gtsam.Ordering)
    9. gtsam.gtsam.JacobianFactor(graph: gtsam::GaussianFactorGraph, ordering: gtsam.gtsam.Ordering, p_variableSlots: gtsam::VariableSlots)
    10. gtsam.gtsam.JacobianFactor(factor: gtsam.gtsam.GaussianFactor)

Invoked with: 8646911284551352320, array([[1., 0., 0.],
       [0., 1., 0.],
       [0., 0., 1.]]), array([[0.],
       [0.],
       [0.]]), unit (3) 
```
indicates that Pybind11 thinks the supplied types are not what it expected. I assume you are building with SYSTEM Eigen, so that should be fine. Could you test if you just add one Python test that does a simple
```
import gtsam
jf = gtsam.JacobianFactor()
```
and see if that works?

If that works, it's the dynamic casting between `gtsam::noiseModel::Gaussian` and its derived classes.

dkogan, 2025-05-07:
Hi. That does work.

ProfFan, 2025-05-08:
Ah, UTM ARM32 emulation is super slow, I just got the Debian installed but didn't have time to debug

dkogan, 2025-05-12:
Hi. I worked on this for a bit. Clearly the python bindings are being confused about what types are compatible: for an arbitrary python function with errors I printed out the types of `self` and of all the inputs, and they clearly match what the error message say should be accepted. I.e. we're giving it correct types, but they're being rejected anyway.

Are you able to debug this? You shouldn't need "emulation" to run arm32 code on arm64 (or i386 on amd64). I can help you reproduce this easily, if that's useful.

ProfFan, 2025-05-23:
Sorry about the late response - I am still attending ICRA and will get back home tomorrow. I have already debugged this for a few hours and I have a rough idea where the problem is, but I need a few more hours to be sure and make a fix.

ProfFan, 2025-06-02:
QEMU x86 on M4 mac is too slow, I got a new 9950x3d, hope it could be faster

ProfFan, 2025-06-04:
I think I know what is wrong, it's simple and stupid: 32 bit architectures we should use 32 bit `Key`s and the mask for the `Symbol` calls should also be set to 32-bit ranges

ProfFan, 2025-06-04:
Need to update `wrap` to generate the correct type for keys

ProfFan, 2025-06-04:
`.def(py::init<size_t, gtsam::Matrix, gtsam::Vector, double>(), py::arg("j"), py::arg("G"), py::arg("g"), py::arg("f"))`

This `size_t` should be `Key`

ProfFan, 2025-06-05:
@dkogan I pushed a PR that fixes this problem, however I haven't got time to do the same to the 4.2 release branch. If you can adapt the PR to 4.2 it would be great, but if you don't have time I can do that later.

I found another issue which is `slam.i` is a bit too big and exhausted the 4GB max memory of 32-bit processes. For 4.2 it's still good but for `develop` it OOMs.

ProfFan, 2025-06-07:
@dkogan What is the deadline for trixie? I can work on this this weekend.

ProfFan, 2025-06-07:
Yeah ideally the best solution is a split of `gtsam` (32/64) and `gtsam-python` (only 64)...

ProfFan, 2025-06-07:
It's a bit too late today and I fixed most of the 32 bit Python issues. However we still have 18 Python unit tests that break non-trivially. So let's keep this issue open and just keep GTSAM 4.2 in Debian 64-bit only.